RIOT-OS/RIOT #17770: Whitespace-in-CFLAGS regression in docker builds

chrysn, 2022-03-08:
#### Description

Since https://github.com/RIOT-OS/RIOT/pull/17396, not all cases of whitespace in CFLAGS can be moved into Docker any more.

#### Steps to reproduce the issue

In any example (I tried examples/rust-gcoap), add this line to the Makefile:

```
CFLAGS += -DGCOAP_STACK_SIZE='(THREAD_STACKSIZE_DEFAULT + DEBUG_EXTRA_STACKSIZE \
    + sizeof(coap_pkt_t) + 3072)'
```

and run the test with and without BUILD_IN_DOCKER=1.

#### Expected results

Both should work.

#### Actual results

With BUILD_IN_DOCKER=1, I get `/bin/sh: 1: Syntax error: "(" unexpected`.

#### Workaround

Putting all in one line and removing the spaces, everything works.

Pinging @fjmolinas who found the right commit that caused the regression.

fjmolinas, 2022-03-17:
I think it might be best simply not to pass CFLAGS do docker directly then, otherwise, everything needs to be escaped (see https://github.com/RIOT-OS/RIOT/blob/master/tests/build_system_cflags_spaces/README.md#setting-cflags-with-space-for-docker). And it just seems a bit of a hassle to do in Make....

fjmolinas, 2022-04-20:
Fixed with https://github.com/RIOT-OS/RIOT/pull/17818

closed 2022-04-20 by fjmolinas