RIOT-OS/RIOT #17788: adc_noise: the LSB of the SAML21 ADC is always zero.

jue89, 2022-03-10:
According to the [chip's errata (Section 1.8.1)](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM_L21_Family_Errata_DS80000781B.pdf), the LSB is not usable as entropy source. Despite the errata suggests to use 12-bit mode to get non-zero bits, that doesn't work with floating inputs, too.

This non-functioning LSB is used as an entropy source:

https://github.com/RIOT-OS/RIOT/blob/cd1c64c8ff429a5a41db4910cda78dc443db6605/sys/entropy_source/adc_noise/adc_noise.c#L39

This situation leads to the test `test/pkg_mbedtls` failing on the `samr30` board. (The `saml21` has a hwrng.)

Does anybody have an idea how to obtain entropy on this chip?

I started using the second bit, which works for my use case, as we have a noisy analog sensor attached to the input. But I'm not sure if this is a good solution for all cases ...

dylad, 2022-03-10:
> Does anybody have an idea how to obtain entropy on this chip?

Probably through the embedded RF chip.
IIRC, Microchip (Atmel) decided to remove the HWRNG from SAML21 in this family because the AT86RF212B already has something "equivalent"

PeterKietzmann, 2022-04-21:
https://doc.riot-os.org/group__sys__puf__sram.html

PeterKietzmann, 2022-07-26:
Can we close this?

closed 2022-07-27 by jue89