RIOT-OS/RIOT #17845: z1(msp430): ztimer_msec is too slow

fjmolinas, 2022-03-22:
#### Description

Compared to `ztimer_usec` `ztimer_msec` on `z1` is 6 times slower for `ztime_set`, and almost 10 times slower for `ztimer_now`. It is not clear where this comes from but probably `ztimer_convert_frac` since that is the main diference.

#### Steps to reproduce the issue

Run `tests/bench_ztimer` with the different backedns:

- usec
```
CFLAGS=-DZTIMER=ZTIMER_USEC BOARD=z1 make -C tests/bench_ztimer/ flash test -j
```

- msec
```
BOARD=z1 make -C tests/bench_ztimer/ flash test -j
```

#### Expected results

No more than ~x2 overhead;

#### Actual results

- usec
```
                     set() one    75134 / 1000 = 75
                  remove() one     9822 / 1000 = 9
          set() + remove() one   128896 / 1000 = 128
  set() many increasing target    22602 / 100 = 226
               re-set()  first   492829 / 1000 = 492
               re-set() middle   494256 / 1000 = 494
               re-set()   last   487628 / 1000 = 487
       remove() + set()  first   543417 / 1000 = 543
       remove() + set() middle   542216 / 1000 = 542
       remove() + set()   last   539568 / 1000 = 539
      remove() many decreasing    11782 / 100 = 117
                  ztimer_now()    17517 / 1000 = 17
              sizeof(ztimer_t)     1000 / 100 = 10

```
- msec
```
                     set() one   469110 / 1000 = 469
                  remove() one     8842 / 1000 = 8
          set() + remove() one   912485 / 1000 = 912
  set() many increasing target    43961 / 100 = 439
               re-set()  first   471125 / 1000 = 471
               re-set() middle   471182 / 1000 = 471
               re-set()   last   471238 / 1000 = 471
       remove() + set()  first   919610 / 1000 = 919
       remove() + set() middle   919735 / 1000 = 919
       remove() + set()   last   919611 / 1000 = 919
      remove() many decreasing    51409 / 100 = 514
                  ztimer_now()   208456 / 1000 = 208
              sizeof(ztimer_t)     1000 / 100 = 10

```
- usec64
```
                     set() one   174819 / 1000 = 174
                  remove() one     7399 / 1000 = 7
          set() + remove() one   181448 / 1000 = 181
  set() many increasing target    30257 / 100 = 302
               re-set()  first   596820 / 1000 = 596
               re-set() middle   597118 / 1000 = 597
               re-set()   last   594431 / 1000 = 594
       remove() + set()  first   718652 / 1000 = 718
       remove() + set() middle   719723 / 1000 = 719
       remove() + set()   last   719585 / 1000 = 719
      remove() many decreasing    18094 / 100 = 180
                  ztimer_now()    24393 / 1000 = 24
              sizeof(ztimer_t)     1400 / 100 = 14
done.
```

maribu, 2023-05-17:
@kaspar030: Do we consider this an issue? IMO we don't have any guarantees that different clocks have similar overhead, so this is IMO fine. In fact, we could easily expose an I2C RTC such as an DS1307 as a ztimer clock with a bit of glue code, which naturally would involve more overhead than a peripheral high frequency timer. I like this flexibility.

maribu, 2023-05-17:
Let me close this for now. If anyone disagrees, please re-open.

closed 2023-05-17 by maribu