RIOT-OS/RIOT #17869: [Tracking] Possible (Research) Projects in RIOT

benpicco, 2022-03-26:
The recent [forum thread](https://forum.riot-os.org/t/20-additional-engineers-dedicated-to-riot-maintenance-development/3574) about what we would do if we had significantly more engineering resources sparked many good ideas, with many sensible suggestions to improve the *existing* code base.

There are always some people who are doing a research project / thesis related to RIOT, there cleanups and refactoring are not very exciting proposals.
So here are some *juicy* topics that I could think of, which, while ambitious, could also be good way to get into RIOT development.

Feel free to edit this post if you can come up with more ideas / details.

## 'Fuzzy' Timers

Add an API to schedule a wake up with a min / max deadline. RIOT should try to coalesce timeouts that fall into the same interval to reduce overall wake-up events. 

## Multiprocessor Support

No SMP as MCUs with multiple cores are not always symmetric (e.g. only the network core can access the radio registers).
Add a mechanism to assign threads to a CPU core and bring up the additional cores.

## Secure wireless networks

IEEE 802.15.4 MAC layer security exists, but is only the first building block for a secure network.
It relies on a shared key that needs to be exchanged on network join and that should be exchanged periodically.

A certificate mechanism for nodes to authenticate themselves to the coordinator could be imagined.

## Porting RIOT to a new MCU family / architecture

This is always an option as there is still plenty of hardware not supported by RIOT. 
There is a [forum thread](https://forum.riot-os.org/t/unsupported-cpus-boards/71/) with some suggestions, but there are always new MCUs being released and many vendors not supported by RIOT at all.

wosym, 2022-07-03:
FYI: first link is not publicly accessible. (Not even with an account)

shibcreate, 2025-08-11:
Sounds good, thank you for the information! I will take a look.