RIOT-OS/RIOT #17907: semtech_loramac_init blocking

ge-lem, 2022-04-07:
#### Description

the function  semtech_loramac_init is blocking, never return, in the example RIOT/examples/lorawan

https://github.com/RIOT-OS/RIOT/blob/a687552ac9d1c50cc9a9d7b3df7a880e9690f3e2/examples/lorawan/main.c#L165

Working in version 2022.01

#### Steps to reproduce the issue

Run the example RIOT/examples/lorawan with a board with sx1276. You dont need to set APPKEY and DEVEUI to test.

The program is stuck at semtech_loramac_init 
https://github.com/RIOT-OS/RIOT/blob/a687552ac9d1c50cc9a9d7b3df7a880e9690f3e2/examples/lorawan/main.c#L165

#### Versions

Working in version 2022.01

geonnave, 2022-04-18:
Hi, you could try enabling debug messages in the drivers to see what's going on.

For example, in `pkg/semtech-loramac/contrib/semtech_loramac.c`:

```diff
-#define ENABLE_DEBUG 0
+#define ENABLE_DEBUG 1
```

aabadie, 2022-04-20:
Can you check if your problem is fixed by disabling the EEPROM usage ? (by trying PR #17969)

fjmolinas, 2022-04-21:
Re-opening since there is no confirmation that this is actually fixed yet, it was auto-closed by the merged #17970

maribu, 2023-01-05:
Let's say the absence of further complaints is confirming that https://github.com/RIOT-OS/RIOT/pull/17970 indeed fixed the issue :)

closed 2023-01-05 by maribu

ant9000, 2023-01-21:
Sorry to break bad news, but I've stumbled on this problem right now on a SAMR34-XPRO board.

I'm testing `examples/lorawan` on master branch, with `ENABLE_DEBUG=1` in the following files:

contrib/semtech_loramac.c
contrib/semtech_loramac_getset.c
contrib/semtech_loramac_radio.c

Here is an excerpt of what happens at board reset:
```
2023-01-21 16:57:43,706 # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2023-01-21 16:57:48,585 # [semtech-loramac] initializing loramac
2023-01-21 16:57:48,621 # [semtech-loramac] unexpected netdev event received: 8
2023-01-21 16:57:48,626 # [semtech-loramac] radio: initialization successful
2023-01-21 16:57:48,665 # [semtech-loramac] set dr 0
2023-01-21 16:57:48,668 # [semtech-loramac] set adr 0
2023-01-21 16:57:48,671 # [semtech-loramac] set public network 1
2023-01-21 16:57:48,674 # [semtech-loramac] set class 0
2023-01-21 16:57:48,679 # main(): This is RIOT! (Version: 2023.04-devel-162-g90c2d)
2023-01-21 16:57:48,683 # LoRaWAN Class A low-power application
2023-01-21 16:57:48,686 # =====================================
2023-01-21 16:57:48,689 # [semtech-loramac] set dr 5
2023-01-21 16:57:48,691 # Starting join procedure
2023-01-21 16:57:48,695 # [semtech-loramac] Starting join procedure: 0
2023-01-21 16:57:48,698 # [semtech-loramac] loramac cmd msg
2023-01-21 16:57:48,701 # [semtech-loramac] starting OTAA join
2023-01-21 16:57:48,703 # [semtech-loramac] get dr
```

No radio packet is emitted (at least, nothing arrives at my gateway), and the board is stuck.