RIOT-OS/RIOT #17931: cpu/stm32/periph/gpio: Delayed last external GPIO interrupt on STM32f767zi

ststrell, 2022-04-13:
#### Description

Using a Nucleo-f767zi with STM32f767zi, the last callback function call of a row of interrupts on a GPIO is delayed by >10ms after the edge on the pin. Interrupts before are all fine. 

FYI @maribu 

#### Steps to reproduce the issue

Input pin configuration:

```c
void isr_callback_input_pin(void *_dev)
{
     gpio_toggle(GPIO_PIN(PORT_B, 9));
}

const gpio_flank_t detectedEdges = GPIO_BOTH;
const gpio_t inputPin = GPIO_PIN(PORT_F, 14);

gpio_init_int(inputPin, GPIO_IN, detectedEdges, isr_callback_input_pin, NULL);

```

Externally toggle input pin several times e.g. > 50 and monitor the input vs output pin state. I did not try it with less edges yet, maybe this problem also applies there.
#### Expected results

Call ISR callback immediately after pin state changes as done with the previous edges.

#### Actual results

Callback is delayed by >10 ms. The screenshot of the logic analyzer shows this behavior. Channel D3 is connected to the input pin of cpu + logic analyzer. Channel D7 is connected to an output pin, which gets toggled for every ISR callback function call. An edge on D3 should therefore be followed by an edge on D7. This works for all previous edges but not for the last one.

```
Receiver Output ------- > CPU input pin
                ------- > Channel D3 logic analyzer
CPU ISR out pin ------- > Channel D7 logic analyzer   
```
[image]

#### Versions

maribu, 2022-06-13:
@ststrell I still didn't have time to look into this. I recall that you mentioned some issue with printf being used in an untimely spot. Did this cause the issue here and it can be closed? Otherwise, I would try to allocate some time on this soonish.

ststrell, 2022-06-13:
@maribu  yeah, it seems like it was my mistake. Currently it works fine, and we can close this issue.

closed 2022-06-13 by ststrell