RIOT-OS/RIOT #17972: stdio_ethos: infinit shell loop

fjmolinas, 2022-04-21:
#### Description

Since https://github.com/RIOT-OS/RIOT/pull/17265 usage of `stdio_ethos` will result in a continuously exiting shell and therefore a barrage of prompts with the shell unusable.

This issue is solved if rebooting (hardware reboot) the device, but it renders automatic tests like `examples/suit_update` impossible.

#### Steps to reproduce the issue

Flash an applicationn using `stdio_ethos`, e.g.: `examples/suit_update`, `examples/gnrc_border_router`.

#### Expected results

Shell boots up fine.

#### Actual results

```
main(): This is RIOT! (Version: 2022.07-devel-135-gba15a6-pr-17950)
RIOT SUIT update example application
Running from slot 0
Image magic_number: 0x544f4952
Image Version: 0x6261064b
Image start address: 0x00002400
Header chksum: 0xe5ca2a4e

suit_coap: started.
Starting the shell
> > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > > 
```

#### Versions

```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04.1 LTS (Focal Fossa)"
                   Kernel: Linux 5.13.0-1017-oem x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: Ubuntu clang version 13.0.0-2

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 4.2.1
                    cmake: cmake version 3.18.4
                 cppcheck: Cppcheck 2.3
                  doxygen: 1.9.1
                      git: git version 2.32.0
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.11.0+dev-00282-gae6de2f93-dirty (2021-11-19-15:06)
                   python: Python 3.9.7
                  python2: Python 2.7.18
                  python3: Python 3.9.7
                   flake8: 4.0.1 (mccabe: 0.6.1, pycodestyle: 2.8.0, pyflakes: 2.4.0) CPython 3.9.7 on
               coccinelle: spatch version 1.1.1-00068-g49826315 compiled with OCaml version 4.08.1
```

miri64, 2022-04-21:
Brainstorming for solutions: I am somewhat wondering how https://github.com/RIOT-OS/RIOT/pull/17265 could have caused this... the shell and its prompt is controlled by the `shell` module, so there is at least some layers of abstraction between the bug and the cause: `stdio` but also the `stdio` handler of `ethos`. So why is this happening?

miri64, 2022-04-21:
> Flash an application using `stdio_ethos`, e.g.: `examples/suit_update`, `examples/gnrc_border_router`.

This does not seem to be the case in general. The changes in #17265 worked fine on the IoT-LAB M3 (and at least @chrysn and @benpicco do not seem to encounter this issue either). Which boards were you using for that.

fjmolinas, 2022-04-21:
> This does not seem to be the case in general. The changes in #17265 worked fine on the IoT-LAB M3 (and at least @chrysn and @benpicco do not seem to encounter this issue either). Which boards were [...]

`nrf52840-mdk`, `dwm1001` the ones I have with me right now...

I also saw it on `iotlab-m3` a couple of times but quite rate, I don't think I saw it on `samr21-xpro`...

miri64, 2022-05-23:
As I am unable to reliably reproduce this on boards I have at hand at the moment (or rather not at all), I think some remote debugging is required... From the output you provided, it looks to me that the shell is running into this condition:

https://github.com/RIOT-OS/RIOT/blob/37db22edeafcb7a4f6190ea5b93bf35128f4bb73/sys/shell/shell.c#L490-L491

This is caused by either EOT or EOF sent by `stdin`

https://github.com/RIOT-OS/RIOT/blob/37db22edeafcb7a4f6190ea5b93bf35128f4bb73/sys/shell/shell.c#L430-L434

So if this bug is indeed `stdio_ethos` related, those characters (`\x04` and `\xff`) should show up after this function call:

https://github.com/RIOT-OS/RIOT/blob/37db22edeafcb7a4f6190ea5b93bf35128f4bb73/drivers/ethos/stdio.c#L70

Could you maybe add a check like this:

```C
if ((tmp > 0) && ((*ptr == 0xff) || (*ptr == 0x04))) {
    /* do something (e.g. toggle LED) */
}
```

Then at least we would see if the issue is from here.

miri64, 2022-05-25:
Can reproduce with `feather-nrf52840dk`! Will try to debug.

jia200x, 2022-05-25:
I have been running into this problem lately relative often. If this helps, I've seen it happening under this conditions:
- Rebooting the device while ethos is running
- It seems to be triggered when nodes are printing to stdout and transmitting frames through netdev_tap

In my use case I have an nrf52850dk gateway running ethos and a DSME interface. I forward all DSME traffic through ethos but at the same time I print some MAC metrics when scheduling/transmitting a frame. In order to avoid the issue, I need to set some timers between commands.

I might be wrong, but I think I've seen these kind of issues even before https://github.com/RIOT-OS/RIOT/pull/17265, but since most of the times ethos run on a samr21-xpro (or equivalent), we probably didn't catch it before.

miri64, 2022-05-25:
> In my use case I have an nrf52850dk gateway running ethos and a DSME interface. I forward all DSME traffic through ethos but at the same time I print some MAC metrics when scheduling/transmitting a frame. In order to avoid the issue, I need to set some timers between commands.

For me, just hitting the on-board reset button also just stops it from happening.

Digging a bit through the code, it seems `_read_r` in `sys/newlib_syscalls_default/syscalls.c` only seems to be called once. After that, `getchar` just always returns `EOF`.

miri64, 2022-05-25:
Was able to debug `getchar` down to

```
Thread 2 hit Breakpoint 2, __srget_r (ptr=ptr@entry=0x20000274 <impure_data>, fp=fp@entry=0x20005b74) at ../../../../../../../../newlib-4.2.0.20211231/newlib/libc/stdio/rget.c:40
40	../../../../../../../../newlib-4.2.0.20211231/newlib/libc/stdio/rget.c: No such file or directory.
Recursive internal problem.
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_border_router/../../Makefile.include:884: debug] Terminated
Terminated
```

:confused:

miri64, 2022-05-25:
> Now I suddenly can't reproduce anymore astonished

back in business... appearently `make flash term` seems to cause it, while "make flash; sleep 1; make term" seems to work (sometimes).

jia200x, 2022-05-25:
> back in business... appearently make flash term seems to cause it, while "make flash; sleep 1; make term" seems to work (sometimes).

Same here! I usually trigger a `make reset` before `make term` in the experiment scripts to avoid that.

miri64, 2022-05-25:
But this seems to be more of a `newlib`+`stdio` problem than a `ethos_stdio` issue. I suspect (but could not show yet) that the `FILE` objects for the re-entrants are somewhat broken, whenever `newlib` calls [CHECK_INIT()](https://www.sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/stdio/getc.c;h=c8781320e60327dcb0e041519eccfe9f4f0c3d3f;hb=484d2ebf8d825b28af47b8ee88cd845f1d2c7c6e#l83)

miri64, 2022-05-25:
Using picolib also seems to work

miri64, 2022-05-25:
Grr nope with that I get a flaky `stdin`...

```
> h 
help
el
lp
p

Command              Description
---------------------------------------
reboot               Reboot the node
version              Prints current RIOT_VERSION
pm                   interact with layered PM subsystem
ps                   Prints information about running threads.
ping6                Ping via ICMPv6
ping                 Alias for ping6
nib                  Configure neighbor information base
ifconfig             Configure network interfaces
6ctx                 6LoWPAN context configuration tool
```

(literally had to type `h<enter>e<enter>l<enter>p<enter>` to get that output...

jia200x, 2022-05-25:
if you connect to /dev/ttyACM0 directly after it gets in the loop (instead of using ethos), does it also print endless ">"?

miri64, 2022-05-25:
> if you connect to /dev/ttyACM0 directly after it gets in the loop (instead of using ethos), does it also print endless ">"?

If I do

```console
$ PROGRAMMER=jlink BOARD=feather-nrf52840 PORT=/dev/ttyUSB0 TERMPROG=picocom TERMFLAGS="--nolock --imap lfcrlf --baud 115200 /dev/ttyUSB0" make -C examples/gnrc_border_router -j flash term
```

no.

miri64, 2022-05-25:
Seems to be whatever was last entered into the terminal (even after a restart of `make term`).

miri64, 2022-05-25:
Ok, following the trace doesn't seem to make any sense (most likely due to optimization?). Going down the rabbit hole, I land [here](https://www.sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/stdio/refill.c;h=ccedc7af72c47f7a4c354509a913c67d13ec49df;hb=484d2ebf8d825b28af47b8ee88cd845f1d2c7c6e#l90) at some point according to gdb in `__srefill_r`.

```C
      /*
       * We were reading.  If there is an ungetc buffer,
       * we must have been reading from that.  Drop it,
       * restoring the previous buffer (if any).  If there
       * is anything in that buffer, return.
       */
      if (HASUB (fp))
	{
	  FREEUB (ptr, fp);
	  if ((fp->_r = fp->_ur) != 0)
	    {
	      fp->_p = fp->_up;
	      return 0;
	    }
	}
```

resulting in `__sgetc_r()` to return -1 [here](https://www.sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/stdio/getc.c;h=c8781320e60327dcb0e041519eccfe9f4f0c3d3f;hb=484d2ebf8d825b28af47b8ee88cd845f1d2c7c6e#l100).

```C
int
getc (register FILE *fp)
{
  int result;
  struct _reent *reent = _REENT;

  CHECK_INIT (reent, fp);
  _newlib_flockfile_start (fp);
  result = __sgetc_r (reent, fp);
  _newlib_flockfile_end (fp);
  return result;
}
```

However, this doesn't make any sense, as both as a pre- and post-condition to `__srefill_r` neither is `fs->_ub.base != NULL` (the condition under which `HASUB(fp)` is true, nor is `(fp->r = fp->_ur) != 0`.

All in all, I am pretty much at a loss what might be the problem. Given that this error happens so deep into `newlib`'s internals, I tend to agree with @jia200x's [comment](https://github.com/RIOT-OS/RIOT/issues/17972#issuecomment-1137033356): https://github.com/RIOT-OS/RIOT/pull/17265 seems not to be the cause, it just made it more obvious due to the performance increase.

fjmolinas, 2022-06-02:
> Using picolibc also seems to work as a workaround:

Sadly not a workaround I can use with BLE border routers, since for some reason NimBLE does not work with Picolibc (at least not the border router)