RIOT-OS/RIOT #17986: CoAP Resource Discovery using query parameters

ad6sh, 2022-04-22:
#### Description
Using CoAP resource discovery using multicast does not  work as expected.
Trying to use the resource discovery as explained in [IETF documentation](https://www.ietf.org/archive/id/draft-ietf-core-resource-directory-28.html#name-a-simple-example) does not work in riot.

sending CoAP multicast GET request to the /.well-known/core with query parameters  is way to discovery as mentioned in the doc.
So as a first step, sending coap multi cast to  /.well-known/core  was tested. 
Then tried adding query parameters to the URI. For eg something like  /.well-known/core?rt=temp

#### Steps to reproduce the issue
Using the [gcoap example ](https://github.com/RIOT-OS/RIOT/tree/2022.01-branch/examples/gcoap),client and server runs on the native instances and they communicate via tap interface,2 issues were noted.

1. **coap get ff02::1 5683 /.well-known/core**  command produces time out. This issue can be solved by the PR[ here](https://github.com/RIOT-OS/RIOT/pull/17978) 

2. Next issue was when adding query parameters to the request.For eg. using the command  **coap get ff02::1 5683 /.well-known/core?rt=temp** .Server returns a 4.04 in this case.
 [gcoap_req_init()](https://doc.riot-os.org/group__net__gcoap.html#ga291fd969b90b9571118c5739706b0c50) takes the path argument as last parameter and **/.well-known/core?rt=temp**  is treated as  a path, thus the URI that’s being accessed in the server is   **/.well-known/core?rt=temp**, which indeed is a path that’s unavailable.

So now its not possible to add query parameters during discovery.

@chrysn

chrysn, 2022-04-22:
> thus the URI that’s being accessed in the server is /.well-known/core?rt=temp, which indeed is a path that’s unavailable.

Actually in URI syntax it's /.well-known/core%3Frt=temp -- the Uri-Path option becomes "core?rt=temp".

Proposed resolution: rename the path argument everywhere from the parameter in the gcoap example down to gcoap_req_init to "path_and_query" (or "path_etc"), and implement setting both. I consider it unlikely that anyone actually relied on a question mark being passed into a path. (I'd be reluctant to call it "URI reference" because a) it only takes URI references that start with a slash, and b) we don't process all URI references, just the subset that kinda probably works without much of a hassle).

ad6sh, 2022-04-30:
Hi @chrysn ...as a first step,i was trying to use the [coap_opt_add_uri_query()](https://doc.riot-os.org/group__net__nanocoap.html#ga05535cd5dff6273ca1dab93844d73b54) from nanocoap.But this functionality does not seems to work in my case.

I have tried to add the query paramerts in the gcoap example itself,to check the functionality.[In this example](https://github.com/ad6sh/coap_resource_discovery_test),you can see a command "disc" which specifically tests this function.

For testing purposes I have started two RIOT CoAP servers.The resources in server has different link parameters.For eg.**rt=xyz** and **rt=abc**. On the client side I use `    coap_opt_add_uri_query(&pdu, "rt", "abc");
` to specifically query from one server only. It doesn't matter whatever query parameters that I have added. The code works like there is no query parameters.Sometimes I get reply from one server and sometimes from another.You could test this by running two server and one client ,then typing the command "disc" on the client several times.

I have also tried to remove the link parameters from server and adding query parameters in client.Even then the client gets reply from server.

I could see [some examples](https://github.com/RIOT-OS/RIOT/blob/73ccd1e2e721bee38f958f8906ac32e5e1fceb0c/sys/net/application_layer/cord/ep/cord_ep.c) using query params in RIOT. And [my implementation](https://github.com/ad6sh/coap_resource_discovery_test/blob/fc8848b579850f18239028d481da9dd66bcabbb1/client.c#L400)  is also exactly like this.

what could be the issue ?(I am not sure If I post this here or at the [RIOT FORUM](https://forum.riot-os.org/t/coap-resource-discovery/3599))

chrysn, 2022-09-21:
Revisiting this, this is a bunch of different issues.

Let's start them off one after the other (to be extended):

* https://github.com/RIOT-OS/RIOT/issues/18622