RIOT-OS/RIOT #18060: Why have `unused-function` compile-times error?

kfessel, 2022-05-05:
#### Description

Since Riot is a library I don't think a unused-function is something to warn about.
Solving for this Warning / Error leadsto:
- more ifndefs (no one likes these) or
- to temporary disabling this warning or 
- to functions that could have been static but are made nonstatic just for the sake of silencing this misguiding message

#### Steps to reproduce the issue

write some static function don't use it and compile. 

#### Expected results

no Warning / Error just a binary that not execs that code

#### Actual results

crying compiler (Error you didn't use all the things)

#### Versions

valentinpi, 2022-06-22:
RIOT is not a library, but a kernel, but I get where you are coming from. Why not add
```Makefile
CFLAGS += -Wno-unused-variable
```
To your Makefile? (not tested)

kfessel, 2022-06-22:
Riot is a library OS

`CFLAGS += -Wno-unused-variable` is the way to disable the warning

I think this would be a good default for evry riot compile since it avoids 'more ifndefs (no one likes these)' and 'functions that are made nonstatic, but should have been static, just for the sake of silencing this misguiding message'

maribu, 2022-08-30:
Note that the compiler only warns if there is absolutely no reference to a function / variable after the preprocessor run left *and* the variable / function is not accessible to external code via a symbol. If you go for

```C
    if (IS_USED(MODULE_FOO)) {
        optional_function();
    }
```

no warning is emitted. So if you avoid preprocessor conditionals as much as possible, this warning rarely pops up. There also is `__attribute__((unused))` that can be added if the call to the function really has to be guarded by preprocessor conditionals.

Generally, it is a good default to warn about functions that do not have any reference at all. Without this warning, `static` functions and variables that become unused after refactoring would occasionally not be removed. Those artifacts would accumulate over time and cause more pain than just adding a `__attribute__((unused))` in the rare cases no `if (IS_USED(...))` is possible.

closed 2022-08-30 by maribu