RIOT-OS/RIOT #18070: flashing: openocd fails to flash the firmware on nucleo-l476rg

Ciusss89, 2022-05-08:
#### Description
Cannot flash the target (a nucleo-l476rg board) due to several errors on the flash procedure. 

#### Steps to reproduce the issue
Basically from the riot-os examples, after compiling the hello-world the flash of target fails due to openocd' errors..
```
~/tuttecose/RIOT/examples/hello-world ((2022.01)) $ make BOARD=nucleo-l476rg flash
Building application "hello-world" for "nucleo-l476rg" with MCU "stm32".

"make" -C /home/giuseppe/tuttecose/RIOT/boards/nucleo-l476rg
"make" -C /home/giuseppe/tuttecose/RIOT/boards/common/nucleo
"make" -C /home/giuseppe/tuttecose/RIOT/core
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/cortexm_common
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/cortexm_common/periph
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32/periph
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32/stmclk
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32/vectors
"make" -C /home/giuseppe/tuttecose/RIOT/drivers
"make" -C /home/giuseppe/tuttecose/RIOT/drivers/periph_common
"make" -C /home/giuseppe/tuttecose/RIOT/sys
"make" -C /home/giuseppe/tuttecose/RIOT/sys/auto_init
"make" -C /home/giuseppe/tuttecose/RIOT/sys/malloc_thread_safe
"make" -C /home/giuseppe/tuttecose/RIOT/sys/newlib_syscalls_default
"make" -C /home/giuseppe/tuttecose/RIOT/sys/pm_layered
"make" -C /home/giuseppe/tuttecose/RIOT/sys/stdio_uart
   text	   data	    bss	    dec	    hex	filename
   8844	    112	   2312	  11268	   2c04	/home/giuseppe/tuttecose/RIOT/examples/hello-world/bin/nucleo-l476rg/hello-world.elf
/home/giuseppe/tuttecose/RIOT/dist/tools/openocd/openocd.sh flash /home/giuseppe/tuttecose/RIOT/examples/hello-world/bin/nucleo-l476rg/hello-world.elf
### Flashing Target ###
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_assert_srst

Info : clock speed 500 kHz
Info : STLINK V2J30M19 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.242688
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Error executing event examine-end on target stm32l4x.cpu:
/usr/bin/../share/openocd/scripts/mem_helper.tcl:37: Error: wrong # args: should be "expr expression"
in procedure 'mmw' called at file "/usr/bin/../share/openocd/scripts/target/stm32l4x.cfg", line 108
at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 37
Info : starting gdb server for stm32l4x.cpu on 0
Info : Listening on port 46555 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32l4x.cpu       hla_target little stm32l4x.cpu       reset

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
Error executing event examine-end on target stm32l4x.cpu:
/usr/bin/../share/openocd/scripts/mem_helper.tcl:37: Error: wrong # args: should be "expr expression"
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 288
in procedure 'mmw' called at file "/usr/bin/../share/openocd/scripts/target/stm32l4x.cfg", line 108
at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 37
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000730 msp: 0x20000200
Info : device idcode = 0x10076415 (STM32L47/L48xx - Rev 4 : 0x1007)
Info : flash size = 1024kbytes
Info : flash mode : dual-bank
Info : Padding image section 1 at 0x080022fc with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x08002300 .. 0x080027ff
auto erase enabled
wrote 8960 bytes from file /home/giuseppe/tuttecose/RIOT/examples/hello-world/bin/nucleo-l476rg/hello-world.elf in 0.581813s (15.039 KiB/s)

verified 8956 bytes in 0.386218s (22.645 KiB/s)

Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
Error executing event examine-end on target stm32l4x.cpu:
/usr/bin/../share/openocd/scripts/mem_helper.tcl:37: Error: wrong # args: should be "expr expression"
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 288
in procedure 'mmw' called at file "/usr/bin/../share/openocd/scripts/target/stm32l4x.cfg", line 108
at file "/usr/bin/../share/openocd/scripts/mem_helper.tcl", line 37

make: *** [/home/giuseppe/tuttecose/RIOT/examples/hello-world/../../Makefile.include:804: flash] Error 1
~/tuttecose/RIOT/examples/hello-world ((2022.01)) $ 

```
#### Versions
```
~/tuttecose/RIOT ((2022.01)) $ ./dist/tools/ci/print_toolchain_versions.sh 

Operating System Environment
----------------------------
         Operating System: "Fedora Linux" "35 (Workstation Edition)"
                   Kernel: Linux 5.17.5-200.fc35.x86_64 x86_64 x86_64
             System shell: GNU bash, version 5.1.8(1)-release (x86_64-redhat-linux-gnu)
             make's shell: GNU bash, version 5.1.8(1)-release (x86_64-redhat-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 11.3.1 20220421 (Red Hat 11.3.1-2)
        arm-none-eabi-gcc: arm-none-eabi-gcc (Fedora 11.1.0-2.fc35) 11.1.0
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "4.1.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.22.2
                 cppcheck: missing
                  doxygen: 1.9.1
                      git: git version 2.35.3
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.11.0
                   python: Python 3.10.4
                  python2: Python 2.7.18
                  python3: Python 3.10.4
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```
RIOT v2022.01

krzysztof-cabaj, 2022-05-09:
Hi. If you need openocd only for flashing (not debugging) as a workaround you could use other programmer for STM boards.

See PR #17550 for more detailed explanation, but if you add at the begin of the file  _**boards/nucleo-l479rg/Makefile.include**_ these lines:

```
DIR_NAME_AT_REMED = "NODE_L479RG"

PROGRAMMERS_SUPPORTED += cpy2remed
```

And later you can flash program using command:

```
make BOARD=nucleo-l479rg PROGRAMMER=cpy2remed flash
```

Ciusss89, 2022-05-10:
It works, but IMHO this implementation has some troubles:

1. On fedora the NUCLEO board gets mounted on  `/run/media/giuseppe/NODE_L476RG`,  so there is a mismatch between (that excludes `run`) the prefixed path and the current one.
2. Patching the prefix for each distro it's not maintainable

BTW, Why openocd did work anymore? I used it last year and the flashing was fine.

aabadie, 2022-05-10:
How did you install openocd ? Using the package manager or by building from source ?

I'm using an openocd version built from source (master at aad87180586a43500f8af1cf79255c7293bb258b, from October 25, 2021) and I can flash without any problem.

Ciusss89, 2022-05-12:
> How did you install openocd ? Using the package manager or by building from source ?
> 
> I'm using an openocd version built from source (master at aad87180586a43500f8af1cf79255c7293bb258b, from October 25, 2021) and I can flash without any problem.

The `openocd` programmer is coming from the fedora's package manager ([0.11.0](https://openocd.org/openocd-0-11-0-released.html)). 

which Linux distro are you using and which is your stm target? I want to isolate the problem, there are three things involved:
1. My linux-distro (Fedora 35)
2. Openocd flashing script
3. Openocd tool bug.

I'm trying to compile with on old tag (to exclude a regression on 2, but I got another error on the linking phase :facepalm:)
```
~/tuttecose/RIOT/examples/hello-world ((2021.01)) $ make BOARD=nucleo-l476rg clean flash
Building application "hello-world" for "nucleo-l476rg" with MCU "stm32".

"make" -C /home/giuseppe/tuttecose/RIOT/boards/nucleo-l476rg
"make" -C /home/giuseppe/tuttecose/RIOT/boards/common/nucleo
"make" -C /home/giuseppe/tuttecose/RIOT/core
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/cortexm_common
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/cortexm_common/periph
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32/periph
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32/stmclk
"make" -C /home/giuseppe/tuttecose/RIOT/cpu/stm32/vectors
"make" -C /home/giuseppe/tuttecose/RIOT/drivers
"make" -C /home/giuseppe/tuttecose/RIOT/drivers/periph_common
"make" -C /home/giuseppe/tuttecose/RIOT/sys
"make" -C /home/giuseppe/tuttecose/RIOT/sys/auto_init
"make" -C /home/giuseppe/tuttecose/RIOT/sys/malloc_thread_safe
"make" -C /home/giuseppe/tuttecose/RIOT/sys/newlib_syscalls_default
"make" -C /home/giuseppe/tuttecose/RIOT/sys/pm_layered
"make" -C /home/giuseppe/tuttecose/RIOT/sys/stdio_uart
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld:cortexm_base.ld:215: warning: memory region `bkup' not declared
/usr/lib/gcc/arm-none-eabi/11.1.0/../../../../arm-none-eabi/bin/ld:cortexm_base.ld:215: syntax error
collect2: error: ld returned 1 exit status
make: *** [/home/giuseppe/tuttecose/RIOT/examples/hello-world/../../Makefile.include:615: /home/giuseppe/tuttecose/RIOT/examples/hello-world/bin/nucleo-l476rg/hello-world.elf] Error 1
~/tuttecose/RIOT/examples/hello-world ((2021.01)) $ 
```

closed 2022-06-01 by Ciusss89