RIOT-OS/RIOT #18103: tests/socket_zep: test flakey due to "Address already in use"

kaspar030, 2022-05-13:
#### Description

Seen this on CI:

```
Traceback (most recent call last):
  File "/tmp/dwq.0.23292065081720303/dcd3a4a8aadb9373e5c0de616499b01b/tests/socket_zep/tests/01-run.py", line 58, in <module>
    s.bind(("127.0.0.1", zep_params['remote_port']))
OSError: [Errno 98] Address already in use
```

Maybe a random port needs to be used?

#### Steps to reproduce the issue

Don't know how to reproduce, it works most of the time.

#### Expected results

#### Actual results

#### Versions

kaspar030, 2022-06-16:
This pops up a lot on the new CI (probably because it is only using one worker at the moment, so the chance of having socket zep users in parallel is much higher).

@miri64 I think you introduced this? What do you think is best?

IIUC we have these options:

1. use random ports -> simple, but still a chance of failing
2. use defined ports per test -> need to manage a list
3. change to use per-build sockets -> more code changes needed

kaspar030, 2022-06-16:
Looks like these applications are using socket_zep:

```
examples/gcoap
examples/gcoap_dtls
examples/gnrc_border_router
examples/gnrc_networking
tests/gnrc_dhcpv6_client_6lbr
tests/gnrc_netif_ieee802154
tests/gnrc_rpl
tests/ieee802154_hal
tests/ieee802154_submac
tests/socket_zep
```

These are using a `ZEP_PORT_BASE` variable:

```
examples/gcoap
examples/gcoap_dtls
examples/gnrc_border_router
examples/gnrc_networking
tests/ieee802154_hal
tests/ieee802154_submac
```

IIUC, that variable is used by both the tests and is passed to the RIOT native instance via TERMFLAGS.

miri64, 2022-06-17:
Maybe @benpicco could also take a look?

miri64, 2022-06-17:
> Looks like these applications are using socket_zep:
>
> ```
> examples/gcoap
> examples/gcoap_dtls
> examples/gnrc_border_router
> examples/gnrc_networking
> tests/gnrc_dhcpv6_client_6lbr [...]

Mh, the line described in OP is only used in `tests/socket_zep` and `tests/gnrc_netif_ieee802154`, as far as I can see. `gnrc_rpl` uses the `zep_dispatch` tool which might also cause problems. The other tests don't seem to interact with the application via `socket_zep` and either just use it to have another interface on the `native` node or have no tests at all.

IMHO we should isolate the tests (e.g. in a [network namespace](https://www.man7.org/linux/man-pages/man8/ip-netns.8.html)) from each other. For network testing we should not expect testers to know all the intricacies of our CI and tests should be run isolated anyways.

kaspar030, 2022-06-19:
> IMHO we should isolate the tests (e.g. in a [network namespace](https://www.man7.org/linux/man-pages/man8/ip-netns.8.html)) from each other.

that'd be quite a bit of work for implementing this in a safe way on CI :(

miri64, 2022-06-20:
> that'd be quite a bit of work for implementing this in a safe way on CI :(

Would it? IIRC it is just a matter of adding an `ip netns exec` before each `make test`.

miri64, 2022-06-20:
... ok, maybe you also need to create it first ^^" but that could be also done with random names.

miri64, 2022-06-20:
Anyways, unless the tests need upstream internet access (which AFAIK they don't)

```
ip netns add ${SOME_RANDOM_NAME}
ip netns exec ${SOME_RANDOM_NAME} make test
ip netns delete ${SOME_RANDOM_NAME}
```

should suffice

kaspar030, 2022-06-20:
> Would it? IIRC it is just a matter of adding an `ip netns exec` before each `make test`.

This is already in docker, so the container needs *at least* the `NET_ADMIN` capability. Not sure if that provides too much access to the host.

kaspar030, 2022-06-20:
> This is already in docker, so the container needs _at least_ the `NET_ADMIN` capability. Not sure if that provides too much access to the host.

Seems like unless docker's `--network=host` is used, `NET_ADMIN` should be fine!

miri64, 2022-06-20:
> This is already in docker, so the container needs _at least_ the `NET_ADMIN` capability. Not sure if that provides too much access to the host.

Mh speaking of: why, if there is already a docker image, do the tests run all in the same container? Separating the test by docker container would probably even easier than having a container in a container.

[image]

kaspar030, 2022-06-20:
> Mh speaking of: why, if there is already a docker image, do the tests run all in the same container?

Well, a typical murdock job takes 2.7s on a fast worker, and starting a container for each test would add a second there.
Hm, but, if we're using docker-compose already, instead of one worker handling 4 parallel jobs per container, maybe we can use docker's "scale" feature and just start 4 workers each handling only one job. That way the startup price is only paid once per container change. I'll try that!

kaspar030, 2022-06-21:
> Hm, but, if we're using docker-compose already, instead of one worker handling 4 parallel jobs per container, maybe we can use docker's "scale" feature and just start 4 workers each handling only one job. That way the startup price is only paid once per container change. I'll try that!

It looks promising, just setting the docker-compose worker dwq instance to "--jobs 1", then starting the whole thing with `docker-compose up -d --scale worker=N` does it.
Only issue is that now the worker threads all identify themselves as e.g., "beast.1". So I want to somehow make the workers know *which* instance(-number) they are, so they can append that to their hostname.
That is actually also an important step towards automatically spreading the workers across numa nodes. :)

kaspar030, 2023-01-07:
Fixed with  https://github.com/RIOT-OS/murdock-worker/pull/1.

closed 2023-01-07 by kaspar030