RIOT-OS/RIOT #18129: add board for Seeedstudio Wio Terminal

donsez, 2022-05-23:
#### Description

Seeedstudio Wio Terminal is a dev board with an ATSAMD51 MCU and Realtek RTL8720DN module for BLE 5.0 & Wi-Fi 2.4G/5G 

The MCU is the ATSAMD51P19 (SAMD51 is already [supported by RIOT](https://github.com/RIOT-OS/RIOT/tree/master/cpu/samd5x)).

The 320x240 LCD screen is driven by ILI9341 (also [supported by RIOT](https://github.com/RIOT-OS/RIOT/tree/master/drivers/ili9341)).

The onboard IMU is LIS3DHTR (also [supported by RIOT](https://github.com/RIOT-OS/RIOT/blob/master/drivers/lis3dh))

Wio Terminal had also a microphone, a speaker, a light sensor (400-1050nm), an Infrared Emitter (940nm), a microSD Card Slot, 40-PIN GPIO (Raspberry Pi Compatible) and 2 Grove connectors (I2C).

### Useful links

* https://wiki.seeedstudio.com/Wio-Terminal-Getting-Started/

maribu, 2022-08-25:
For the record: I do have one of those boards lying around somewhere with the intent to add board support for it. I just never found time to do so. Maybe nagging me a bit could motivate me ;)

I would certainly be able to review and test, if someone else is faster than me.

donsez, 2022-08-25:
+1 for motivation

oschettler, 2022-10-01:
+1 I'd love to see support for Wio Terminal