RIOT-OS/RIOT #18228: Bug: GPIO actions not working with the Arduino Uno nor the Arduino Nano

valentinpi, 2022-06-17:
#### Description

I cannot toggle LEDs or control my buzzers, because after initializing the proper ports the state of them never changes.

#### Steps to reproduce the issue

1. Take an Arduino Uno or an Arduino Nano. (both v3, ATMEGA328P and ATMEGA328)
2. Plug in the device via a USB cable.
3. Create a small project with the following code and it's makefile:
```c
#include <led.h>
#include <periph/gpio.h>
#include <ztimer.h>

int main(void)
{
    const gpio_t LED_PIN = GPIO_PIN(1, 5);

    while (1) {
        printf("%d\n", gpio_read(LED_PIN));
        LED_TOGGLE(0);

        ztimer_sleep(ZTIMER_SEC, 1);
    }

    return 0;
}
```
```Makefile
APPLICATION = hello-world
RIOTBASE ?= $(CURDIR)/../..
DEVELHELP ?= 0
# See https://github.com/qmk/qmk_firmware/issues/17064
CFLAGS += -Wno-array-bounds

USEMODULE += periph_gpio
USEMODULE += ztimer
USEMODULE += ztimer_sec

QUIET ?= 1

include $(RIOTBASE)/Makefile.include
```
4. Flash the firmware e.g. by (I used the `hello-world` example for this to test the GPIO):
```sh
BOARD=arduino-nano make -j clean all; avrdude -c arduino -b 115200 -p m328p -P /dev/ttyUSB0 -U flash:w:/home/valentin/dev/RIOT/examples/hello-world/bin/arduino-nano/hello-world.hex; BOARD=arduino-nano make -j term
```

#### Expected results

The terminal window outputs zeros and ones in alternation, whilst die LED is blinking each second.

#### Actual results

The terminal window only outputs zeros and the LED stays off.

#### Versions

This also does not work with a small sound buzzer I bought today. The devices are not defect as I have connected them to current directly once and it worked. I suspect RIOT is not toggeling the GPIO ports.

### Additional Notes

- To reproduce, one has to adjust the paths a bit or just put the codei n the `hello-world` example like I did.
- I have mailed Hauke regarding the issue from above. Compilation is currently broken with the newest version of `avr-gcc`, but this is not RIOTs failt.

krzysztof-cabaj, 2022-06-18:
Hi, try first initialize GPIO using `init_gpio`. See this [link](https://doc.riot-os.org/group__drivers__periph__gpio.html).

valentinpi, 2022-06-18:
Hi! I tried that, it still does not work on my board. Also, I saw that it gets initialized automatically: `https://github.com/RIOT-OS/RIOT/blob/master/drivers/periph_common/init_leds.c#L39`.

krzysztof-cabaj, 2022-06-19:
Maybe as a workaround you could simply use  `gpio_toggle` instead of `LED_TOGGLE(0)`. I have Arduino UNO in my University and I will try to take it and looked where is a problem in the following week.

valentinpi, 2022-06-19:
I also tried initializing and toggeling manually. It's very strange, maybe I made some weird mistake in my setup? It would be nice if you could see if you are able to reproduce this.

krzysztof-cabaj, 2022-06-21:
Hi, I tested provided code at arduino-uno - indeed there are some problems with GPIO at this board. In current RIOT version (master) your code with `LED_TOGGLE(0)` and with `gpio_toggle(LED_PIN)`(instead LED_TOGGLE) did not work properly. Moreover, in this version of the RIOT code, program `examples/blinky` did not work, too. What is interesting, when I checked this program in older RIOT version (around end of January 2022) version with `LED_TOGGLE(0)` did not work but with `gpio_toggle(LED_PIN)` works.

Currently, no ideas on my end where is bug. I try to looked at this issue in the following week(s).

valentinpi, 2022-06-25:
Hi, thanks for the effort. I will test the fix when I find some time and if it works I will close this issue. I'm not closing yet since I see only the LED module has been changed, but there were some general GPIO problems the last time I used these boards. That's why I could not control my buzzer.

valentinpi, 2022-06-28:
Fixed with #18265.

closed 2022-06-28 by valentinpi