RIOT-OS/RIOT #18262: samr21-xpro: Weird `printf` issue

valentinpi, 2022-06-25:
#### Description

Usage of the `PRIu8` format specifier in `inttypes.h` yields weird behavior on the `samr21-xpro` board.

#### Steps to reproduce the issue

Take a `samr21-xpro` board, connect it and flash the following firmware:

```c
#include <inttypes.h>
#include <shell.h>
#include <stdio.h>

int main(void)
{
    uint8_t i = 25;
    printf("%u\n", i);
    printf("%" PRIu8 "\n", i);

    uint8_t shell_buf[SHELL_DEFAULT_BUFSIZE];
    memset(shell_buf, 0, SHELL_DEFAULT_BUFSIZE);
    shell_run(NULL, (char *)shell_buf, SHELL_DEFAULT_BUFSIZE);

    return 0;
}
```

```Makefile
APPLICATION = hello-world
BOARD ?= samr21-xpro
RIOTBASE ?= $(CURDIR)/../..
DEVELHELP ?= 0

USEMODULE += ps
USEMODULE += shell
USEMODULE += shell_commands

QUIET ?= 1

include $(RIOTBASE)/Makefile.include
```

#### Expected results

The following output upon launching the serial terminal:

```
25
25
```

#### Actual results

```
25
hu
```

#### Versions

I only tested this for this board while debugging another firmware.

### System Information

```
... ~/d/RIOT (master)> make print-versions                            

Operating System Environment 
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.15.49-1-lts x86_64 unknown
             System shell: GNU bash, Version 5.1.16(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, Version 5.1.16(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 12.1.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 12.1.0
                  avr-gcc: avr-gcc (GCC) 12.1.0
         mips-mti-elf-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 13.0.1

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "4.2.0"
      mips-mti-elf-newlib: missing
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.1.0" ("20220128")

Installed development tools
---------------------------
                   ccache: ccache version 4.6.1
                    cmake: cmake version 3.23.2
                 cppcheck: Cppcheck 2.8
                  doxygen: 1.9.3
                      git: git version 2.36.1
                     make: GNU Make 4.3
                  openocd: Open On-Chip Debugger 0.11.0
                   python: Python 3.10.5
                  python2: Python 2.7.18
                  python3: Python 3.10.5
                   flake8: 4.0.1 (mccabe: 0.7.0, pycodestyle: 2.8.0, pyflakes: 2.4.0) CPython 3.10.5 on
               coccinelle: spatch version 1.1.1 compiled with OCaml version 4.13.1
```

miri64, 2022-06-27:
That's not a bug, but an optimization of `newlib`'s nanospecs. Either, use a different libc, avoid those those format specifiers or the `fmt` module instead of `stdio.h`.

miri64, 2022-06-27:
See also https://github.com/RIOT-OS/RIOT/issues/1891 and others.

closed 2022-06-27 by miri64