RIOT-OS/RIOT #18328: remove leading underscore in log module variable name

david-vankampen, 2022-07-19:
https://github.com/RIOT-OS/RIOT/blob/bafb7d53c5b387d79f391b26be947233887361d8/sys/log/log_color/log_module.h#L80

per[ ISO/IEC 9899, Section 7.1.3](https://www.open-std.org/jtc1/sc22/wg14/www/docs/n1570.pdf), 
[image]
variables that begin with a single underscore and a lower case letter are reserved by the language for file scope use.

kfessel, 2022-07-27:
`_ansi_codes` is only used within the scope of that file -> follows that guideline 

```
RIOT$ rg _ansi_codes
sys/log/log_color/log_module.h
80:static const char * const _ansi_codes[] =
98:    printf("%s", _ansi_codes[level]);
```

but this may as well be put into the the soope of that one function using it

on the other hand it is internal only -- it should be but it is not somehow it made its way into `core/lib/include/log.h`

kfessel, 2023-05-17:
This is fixed with #18425

closed 2023-05-17 by kfessel