RIOT-OS/RIOT #18432: gnrc_tcp send fails

uglybob, 2022-08-10:
#### Description
Cannot send from node to node using gnrc with tcp on transport layer. The server part seems to work just fine, as I can send to it from my PC via gnrc_border_router and receive echo messages. 

#### Steps to reproduce the issue
I'm using a slightly modified [TCP sock API example](https://doc.riot-os.org/group__net__sock__tcp.html) with gnrc on nrf52840dk boards. One of the nodes starts the echo server, the other one runs the client example and tries to send a message to the server. Both nodes can ping each other.

#### Expected results
A connection is established and the echo server replies with the same message.

#### Actual results
The client tries to retransmit a couple of times but nothing is sent. The sniffer doesn't show any traffic either.

```
2022-08-10 15:00:20,565 # tcp send fe80::e8d8:51c5:4b95:4183 12345 hi
2022-08-10 15:00:20,573 # GNRC_TCP: Enter "gnrc_tcp_tcb_init", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp.c(311)
2022-08-10 15:00:20,581 # GNRC_TCP: Leave "gnrc_tcp_tcb_init", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp.c(325)
2022-08-10 15:00:20,588 # GNRC_TCP: Enter "gnrc_tcp_open", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp.c(342)
2022-08-10 15:00:20,597 # GNRC_TCP: Enter "_sched_connection_timeout", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp.c(64)
2022-08-10 15:00:20,605 # GNRC_TCP: Enter "_sched_mbox", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp.c(55)
2022-08-10 15:00:20,612 # GNRC_TCP: Leave "_sched_mbox", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp.c(59)
2022-08-10 15:00:20,621 # GNRC_TCP: Leave "_sched_connection_timeout", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp.c(67)
2022-08-10 15:00:20,629 # GNRC_TCP: Enter "_gnrc_tcp_pkt_build", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(153)
2022-08-10 15:00:20,638 # GNRC_TCP: Leave "_gnrc_tcp_pkt_build", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(274)
2022-08-10 15:00:20,648 # GNRC_TCP: Enter "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(381)
2022-08-10 15:00:20,656 # GNRC_TCP: Enter "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(367)
2022-08-10 15:00:20,665 # GNRC_TCP: Leave "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(374)
2022-08-10 15:00:20,675 # GNRC_TCP: Enter "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(367)
2022-08-10 15:00:20,685 # GNRC_TCP: Leave "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(372)
2022-08-10 15:00:20,694 # GNRC_TCP: Leave "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(449)
2022-08-10 15:00:20,703 # GNRC_TCP: Enter "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(281)
2022-08-10 15:00:20,715 # GNRC_TCP: Info: "Received GNRC_NETAPI_MSG_TYPE_SND.", Func: _eventloop, File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(320)
2022-08-10 15:00:20,723 # GNRC_TCP: Enter "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(68)
2022-08-10 15:00:20,730 # GNRC_TCP: Leave "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(100)
2022-08-10 15:00:20,739 # GNRC_TCP: Leave "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(304)
2022-08-10 15:00:21,688 # GNRC_TCP: Info: "Received MSG_TYPE_RETRANSMISSION.", Func: _eventloop, File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(332)
2022-08-10 15:00:21,697 # GNRC_TCP: Enter "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(381)
2022-08-10 15:00:21,706 # GNRC_TCP: Enter "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(367)
2022-08-10 15:00:21,715 # GNRC_TCP: Leave "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(374)
2022-08-10 15:00:21,725 # GNRC_TCP: Enter "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(367)
2022-08-10 15:00:21,734 # GNRC_TCP: Leave "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(372)
2022-08-10 15:00:21,744 # GNRC_TCP: Leave "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(449)
2022-08-10 15:00:21,752 # GNRC_TCP: Enter "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(281)
2022-08-10 15:00:21,761 # GNRC_TCP: Leave "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(304)
2022-08-10 15:00:21,773 # GNRC_TCP: Info: "Received GNRC_NETAPI_MSG_TYPE_SND.", Func: _eventloop, File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(320)
2022-08-10 15:00:21,780 # GNRC_TCP: Enter "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(68)
2022-08-10 15:00:21,789 # GNRC_TCP: Leave "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(100)
2022-08-10 15:00:23,736 # GNRC_TCP: Info: "Received MSG_TYPE_RETRANSMISSION.", Func: _eventloop, File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(332)
2022-08-10 15:00:23,746 # GNRC_TCP: Enter "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(381)
2022-08-10 15:00:23,755 # GNRC_TCP: Enter "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(367)
2022-08-10 15:00:23,764 # GNRC_TCP: Leave "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(374)
2022-08-10 15:00:23,773 # GNRC_TCP: Enter "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(367)
2022-08-10 15:00:23,783 # GNRC_TCP: Leave "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(372)
2022-08-10 15:00:23,793 # GNRC_TCP: Leave "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(449)
2022-08-10 15:00:23,801 # GNRC_TCP: Enter "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(281)
2022-08-10 15:00:23,810 # GNRC_TCP: Leave "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(304)
2022-08-10 15:00:23,822 # GNRC_TCP: Info: "Received GNRC_NETAPI_MSG_TYPE_SND.", Func: _eventloop, File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(320)
2022-08-10 15:00:23,830 # GNRC_TCP: Enter "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(68)
2022-08-10 15:00:23,837 # GNRC_TCP: Leave "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(100)
2022-08-10 15:00:27,786 # GNRC_TCP: Info: "Received MSG_TYPE_RETRANSMISSION.", Func: _eventloop, File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(332)
2022-08-10 15:00:27,795 # GNRC_TCP: Enter "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(381)
2022-08-10 15:00:27,804 # GNRC_TCP: Enter "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(367)
2022-08-10 15:00:27,813 # GNRC_TCP: Leave "_gnrc_tcp_pkt_get_pay_len", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(374)
2022-08-10 15:00:27,823 # GNRC_TCP: Enter "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(367)
2022-08-10 15:00:27,832 # GNRC_TCP: Leave "_gnrc_tcp_eventloop_sched", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(372)
2022-08-10 15:00:27,841 # GNRC_TCP: Leave "_gnrc_tcp_pkt_setup_retransmit", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(449)
2022-08-10 15:00:27,850 # GNRC_TCP: Enter "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(281)
2022-08-10 15:00:27,859 # GNRC_TCP: Leave "_gnrc_tcp_pkt_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_pkt.c(304)
2022-08-10 15:00:27,871 # GNRC_TCP: Info: "Received GNRC_NETAPI_MSG_TYPE_SND.", Func: _eventloop, File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(320)
2022-08-10 15:00:27,879 # GNRC_TCP: Enter "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(68)
2022-08-10 15:00:27,886 # GNRC_TCP: Leave "_send", File: sys/net/gnrc/transport_layer/tcp/gnrc_tcp_eventloop.c(100)
[... 30 lines cut ...]
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: cc1: fatal error: avr/version.h: No such file or directory
[... cut ...]
                    cmake: cmake version 3.13.4
                 cppcheck: missing
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-03-25-17:31)
                   python: Python 2.7.16
                  python2: Python 2.7.16
                  python3: Python 3.7.3
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing

```

brummer-simon, 2022-08-16:
Hi @uglybob,

From the log itself, it looks like the gnrc_tcp_open call never returns. This is an Indicator that the connection was not established in the first place. Do you have some way to provide a wireshark log or something? This might give me more insight whats happens here.

Cheers Simon

uglybob, 2023-01-05:
hi simon,

sorry for the delay and thanks for the quick reply and your implementation! this turned out to be a misconfiguration in routing. the tcp send works just fine.

closed 2023-01-05 by uglybob