RIOT-OS/RIOT #18437: libstdcpp error on Mac

Impeekay, 2022-08-11:
#### Description

unable to run cpp example given in the repository on Mac.

#### Steps to reproduce the issue

- Clone the repo on a Mac
- move to examples/riot_and_cpp
- run using `gmake BOARD=arduino-uno`

#### Expected results

Successfully build the given example.

#### Actual results

```
❯ gmake BOARD=arduino-uno
There are unsatisfied feature requirements: libstdcpp
/Users/pk/projects/college/cse/RIOT/Makefile.include:998: *** You can let the build continue on expected errors by setting CONTINUE_ON_EXPECTED_ERRORS=1 to the command line.  Stop.
```

#### Versions

macOS Monterey (12.4) - Apple M1 MacBook Pro
GNU Make 4.3

I do know that it's unable to find libstdcpp, but I can confirm that libstdc++ is included in Xcode and also I've tried replacing libstdcpp with libstdc++. Since, I'm new to this, I'm confused by this and I might be missing something In this. Any help would be appreciated.

Thanks.

maribu, 2022-09-14:
Sorry for the long delay.

The issue is actually not related to Mac or Mac OS, but to the AVR platform the Arduino Uno uses. Simply put, there is no libstdc++ for AVR (at least avr-libc doesn't provide it).

This doesn't mean that you cannot use C++. E.g. all the Arduino examples will very much use C++. But you cannot used features provided by the libstdc++.

If you really want to use the libstdc++, you may could take a look at https://github.com/modm-io/avr-libstdcpp which provides both libstdc++ and an STL. Note that with the [package ETL](https://doc.riot-os.org/group__pkg__etl.html) there indeed is a standard template library in RIOT. So `avr-libstdcpp` in combination with the `ETL` may or may not be result in some conflicts.

In any case: In order to run more than a simple hello world on the Arduino UNO, you likely want to use [PR #18148](https://github.com/RIOT-OS/RIOT/pull/18148). Otherwise a lot of `const` strings will be placed into RAM, as no part of the flash is mapped in the data address space on the ATmega MCUs.

Impeekay, 2022-09-16:
Hey @maribu, Thanks for the reply. I just tried out the same example on a linux machine, it compiles without any issues. PFA the same. 

[image]

maribu, 2022-09-19:
I think this can be closed now. If there are still questions open or issues remaining, do not hesitate to ask and please reopen the issue.

closed 2022-09-19 by maribu

Impeekay, 2022-09-21:
Hi @maribu thanks for the reply, and letting me know that the platform I was trying to compile to did not support c++. I now have a example program that I try to compile for Arduino-due. But, this time around it fails because of conflicting declarations because one of my dependencies uses the standard library and also the definition exists in newlib. Is there a different way I should compile this or am I totally doing this wrong?

Please do let me know.

[image]

maribu, 2022-09-22:
It looks like headers for the host system are included. E.g. OpenSSL is not supported by RIOT (and it does not make sense to support OpenSSL on micro controllers), still OpenSSL headers end up being included.

If you do need crypto on microcontollers: There are packages for tinydlts, wolfssl, Mbed TLS. I'm not entirely sure if the Mbed TLS package actually provides TLS, as I only have seen it being used for entropy harvesting.

Impeekay, 2022-09-22:
Thanks for the reply @maribu . It looks like I might not be able to port this onto arduino then without modifying the actual source of the library. The library heavily depends on OpenSSL. So, moving away from it, what's a better alternative for microcontrollers in case I can try tinkering with it?

Thanks,
Pavan