RIOT-OS/RIOT #18456: riotboot_dfu: when the app download completes the device is still in DFU mode

eduazocar, 2022-08-16:
#### Description

I have a little bug using riotboot_dfu. Well in my case, I've trying to download (upload this is confuse with dfu-utils meanings) a example via dfu-util (PROGRAMMER), when the download it's finished. The device doesn't init the application. Because, still in DFU mode, I need only access to DFU mode when start download process, when the process finished the device should be in RUNTIME_MODE the application will be launched using riotboot/slot0 I've used the samr21-xpro to test.
#### Steps to reproduce the issue

builds on `samr21-xpro`
You need to add in `Makefile.include` of `samr21-xpro`board the parameter to support dfu-util ( `PROGRAMMER_SUPPORTED=dfu-util` ). don't worry about if this is looks like forced, was tested this board to execute the èxample/usbus_minimal`  so this is a basic requirement to run the DFU bootloader. 

Well, the next step is build and flash the `bootloader/riotboot_dfu` to install it in the board.
```
make -C bootloaders/riotboot_dfu flash
```
then donwload an application via dfu-util.
```
FEATURES_REQUIRED+=riotboot USEMODULE+=usbus_dfu make -C examples/blinky/ BOARD=samr21-xpro PROGRAMMER=dfu-util all riotboot/flash-slot0
```
#### Expected results
The expected result is that application load after that dfu-util complete the download. This means that device should stay in `Runtime mode` after de flashing via dfu-util.

#### Actual results
When the download is finished, the device goes into DFU mode and the second time the application is downloaded to the board it does not work.

#### Versions
Operating system: Ubuntu 22.04 LTS

dylad, 2022-08-22:
So far, I cannot reproduce this issue on `samr21-xpro`
I directly ran:
`BOARD=samr21-xpro make -C bootloaders/riotboot_dfu flash`
then
`FEATURES_REQUIRED+=riotboot USEMODULE+=usbus_dfu make -C examples/blinky/ BOARD=samr21-xpro PROGRAMMER=dfu-util all riotboot/flash-slot0`
against lastest master. It works out of the box on my side.

Are you using lastest master branch ?
Could you also provide the full log of your commands ?

eduazocar, 2022-08-22:
oh you are right, i forgot give you information about change in samr21-xpro board. You need to add in the Makefile.include the following line to add support to DFU mode. 
```
PROGRAMMER = dfu-util
```

eduazocar, 2022-08-22:
> So far, I cannot reproduce this issue on `samr21-xpro` I directly ran: `BOARD=samr21-xpro make -C bootloaders/riotboot_dfu flash` then `FEATURES_REQUIRED+=riotboot USEMODULE+=usbus_dfu make -C [...]
My last commits are:
```
commit 932ca6abfca447e51fa3d2526d68448dbe3cc9b2 (HEAD -> copy_feature/riotboot_dfu, eduazocar/feature/riotboot_dfu)
Author: eduazocar <eduardoazocar7@gmail.com>
Date:   Mon Aug 15 13:00:28 2022 -0400

    fixed_riotboor_dfu: switching from DFU to Runtime mode

commit 932c58d89f732b86988219d5f17dd3ef6842524d (eduazocar/master, RIOT/master)
Merge: fda230eb7c 4dcb8edcc8
Author: Martine Lenders <m.lenders@fu-berlin.de>
Date:   Fri Jul 1 17:11:13 2022 +0200

    Merge pull request #18289 from nmeum/pr/emcute_deadlock_suback
    
    asymcute: Fix deadlocks in REGACK, SUBACK and UNSUBACK handler

commit fda230eb7c0d0422f85154d45a4ee5fff94c2a4e
Merge: cb5d4f88f9 5a47d6a9b8
Author: Martine Lenders <m.lenders@fu-berlin.de>
```
*Note: ignore the reference branch copy_feature_riotboot, because always i do a backup of PR before to do squash*

dylad, 2022-08-22:
I've added the line to Makefile.include. Still works as expected.

Here is my full log:

```
$ dist/tools/edbg/edbg -t samr21 -e                       

$ make BOARD=samr21-xpro -C bootloaders/riotboot_dfu flash

$ make BOARD=samr21-xpro -C bootloaders/riotboot_dfu flash
make : on entre dans le répertoire « /home/dylad/software/RIOT/bootloaders/riotboot_dfu »
Building application "riotboot_dfu" for "samr21-xpro" with MCU "samd21".

"make" -C /home/dylad/software/RIOT/boards/common/init
"make" -C /home/dylad/software/RIOT/boards/samr21-xpro
"make" -C /home/dylad/software/RIOT/core
"make" -C /home/dylad/software/RIOT/core/lib
"make" -C /home/dylad/software/RIOT/cpu/samd21
"make" -C /home/dylad/software/RIOT/cpu/cortexm_common
"make" -C /home/dylad/software/RIOT/cpu/cortexm_common/periph
"make" -C /home/dylad/software/RIOT/cpu/sam0_common
"make" -C /home/dylad/software/RIOT/cpu/sam0_common/periph
"make" -C /home/dylad/software/RIOT/cpu/samd21/periph
"make" -C /home/dylad/software/RIOT/cpu/samd21/vectors
"make" -C /home/dylad/software/RIOT/drivers
"make" -C /home/dylad/software/RIOT/drivers/periph_common
"make" -C /home/dylad/software/RIOT/sys
"make" -C /home/dylad/software/RIOT/sys/checksum
"make" -C /home/dylad/software/RIOT/sys/div
"make" -C /home/dylad/software/RIOT/sys/event
"make" -C /home/dylad/software/RIOT/sys/fmt
"make" -C /home/dylad/software/RIOT/sys/frac
"make" -C /home/dylad/software/RIOT/sys/luid
"make" -C /home/dylad/software/RIOT/sys/malloc_thread_safe
"make" -C /home/dylad/software/RIOT/sys/newlib_syscalls_default
"make" -C /home/dylad/software/RIOT/sys/riotboot
"make" -C /home/dylad/software/RIOT/sys/stdio_null
"make" -C /home/dylad/software/RIOT/sys/usb/usbus
"make" -C /home/dylad/software/RIOT/sys/usb/usbus/dfu
"make" -C /home/dylad/software/RIOT/sys/ztimer
   text	   data	    bss	    dec	    hex	filename
  11604	     60	   2848	  14512	   38b0	/home/dylad/software/RIOT/bootloaders/riotboot_dfu/bin/samr21-xpro/riotboot_dfu.elf
[INFO] edbg binary not found - building it from source now
CC= CFLAGS= make -C /home/dylad/software/RIOT/dist/tools/edbg
[INFO] edbg binary successfully built!
/home/dylad/software/RIOT/dist/tools/edbg/edbg.sh flash /home/dylad/software/RIOT/bootloaders/riotboot_dfu/bin/samr21-xpro/riotboot_dfu.bin
### Flashing Target ###
Debugger: ATMEL EDBG CMSIS-DAP ATML2127031800011458 01.1A.00FB (S)
Clock frequency: 16.0 MHz
Target: SAM R21G18 (Rev C)
Verification...
at address 0x0 expected 0x00, read 0xff
Error: verification failed
Debugger: ATMEL EDBG CMSIS-DAP ATML2127031800011458 01.1A.00FB (S)
Clock frequency: 16.0 MHz
Target: SAM R21G18 (Rev C)
Programming................................................. done.
Verification................................................. done.
Done flashing
make : on quitte le répertoire « /home/dylad/software/RIOT/bootloaders/riotboot_dfu »

$ FEATURES_REQUIRED+=riotboot USEMODULE+=usbus_dfu make -C examples/blinky/ BOARD=samr21-xpro PROGRAMMER=dfu-util all riotboot/flash-slot1
make : on entre dans le répertoire « /home/dylad/software/RIOT/examples/blinky »
Building application "blinky" for "samr21-xpro" with MCU "samd21".

"make" -C /home/dylad/software/RIOT/boards/common/init
"make" -C /home/dylad/software/RIOT/boards/samr21-xpro
"make" -C /home/dylad/software/RIOT/core
"make" -C /home/dylad/software/RIOT/core/lib
"make" -C /home/dylad/software/RIOT/cpu/samd21
[... 55 lines cut ...]
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1209:7d02
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #1 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 64
Copying data from PC to DFU device
Download	[=========================] 100%        17388 bytes
Download done.
state(5) = dfuDNLOAD-IDLE, status(0) = No error condition is present
Done!
make : on quitte le répertoire « /home/dylad/software/RIOT/examples/blinky »
```

I've erased the whole flash before running the commands again.
Please provide me yours so we can identify if something is wrong on your side.

eduazocar, 2022-08-23:
> 

> I've added the line to Makefile.include. Still works as expected.
> 
> Here is my full log:
> 
> ```
> $ dist/tools/edbg/edbg -t samr21 -e                       
> 
> $ make BOARD=samr21-xpro -C bootloaders/riotboot_dfu flash
> make : on entre dans le répertoire « /home/dylad/software/RIOT/bootloaders/riotboot_dfu »
> Building application "riotboot_dfu" for "samr21-xpro" with MCU "samd21".
> 
> "make" -C /home/dylad/software/RIOT/boards/common/init
> "make" -C /home/dylad/software/RIOT/boards/samr21-xpro
> "make" -C /home/dylad/software/RIOT/core
> "make" -C /home/dylad/software/RIOT/core/lib
> "make" -C /home/dylad/software/RIOT/cpu/samd21
> "make" -C /home/dylad/software/RIOT/cpu/cortexm_common
> "make" -C /home/dylad/software/RIOT/cpu/cortexm_common/periph
> "make" -C /home/dylad/software/RIOT/cpu/sam0_common
> "make" -C /home/dylad/software/RIOT/cpu/sam0_common/periph
> "make" -C /home/dylad/software/RIOT/cpu/samd21/periph
> "make" -C /home/dylad/software/RIOT/cpu/samd21/vectors
> "make" -C /home/dylad/software/RIOT/drivers
> "make" -C /home/dylad/software/RIOT/drivers/periph_common
> "make" -C /home/dylad/software/RIOT/sys
> "make" -C /home/dylad/software/RIOT/sys/checksum
> "make" -C /home/dylad/software/RIOT/sys/div
> "make" -C /home/dylad/software/RIOT/sys/event
> "make" -C /home/dylad/software/RIOT/sys/fmt
> "make" -C /home/dylad/software/RIOT/sys/frac
> "make" -C /home/dylad/software/RIOT/sys/luid
> "make" -C /home/dylad/software/RIOT/sys/malloc_thread_safe
> "make" -C /home/dylad/software/RIOT/sys/newlib_syscalls_default
> "make" -C /home/dylad/software/RIOT/sys/riotboot
> "make" -C /home/dylad/software/RIOT/sys/stdio_null
> "make" -C /home/dylad/software/RIOT/sys/usb/usbus
> "make" -C /home/dylad/software/RIOT/sys/usb/usbus/dfu
> "make" -C /home/dylad/software/RIOT/sys/ztimer
>    text	   data	    bss	    dec	    hex	filename
>   11604	     60	   2848	  14512	   38b0	/home/dylad/software/RIOT/bootloaders/riotboot_dfu/bin/samr21-xpro/riotboot_dfu.elf
> [INFO] edbg binary not found - building it from source now
> CC= CFLAGS= make -C /home/dylad/software/RIOT/dist/tools/edbg
> [INFO] edbg binary successfully built!
> /home/dylad/software/RIOT/dist/tools/edbg/edbg.sh flash /home/dylad/software/RIOT/bootloaders/riotboot_dfu/bin/samr21-xpro/riotboot_dfu.bin
> ### Flashing Target ###
> Debugger: ATMEL EDBG CMSIS-DAP ATML2127031800011458 01.1A.00FB (S)
> Clock frequency: 16.0 MHz
> Target: SAM R21G18 (Rev C)
> Verification...
> at address 0x0 expected 0x00, read 0xff
> Error: verification failed
> Debugger: ATMEL EDBG CMSIS-DAP ATML2127031800011458 01.1A.00FB (S)
> Clock frequency: 16.0 MHz
> Target: SAM R21G18 (Rev C)
> Programming................................................. done.
> Verification................................................. done.
> Done flashing
> make : on quitte le répertoire « /home/dylad/software/RIOT/bootloaders/riotboot_dfu »
> 
> $ FEATURES_REQUIRED+=riotboot USEMODULE+=usbus_dfu make -C examples/blinky/ BOARD=samr21-xpro PROGRAMMER=dfu-util all riotboot/flash-slot1
> make : on entre dans le répertoire « /home/dylad/software/RIOT/examples/blinky »
> Building application "blinky" for "samr21-xpro" with MCU "samd21".
> 
> "make" -C /home/dylad/software/RIOT/boards/common/init
> "make" -C /home/dylad/software/RIOT/boards/samr21-xpro
> "make" -C /home/dylad/software/RIOT/core
> "make" -C /home/dylad/software/RIOT/core/lib
> "make" -C /home/dylad/software/RIOT/cpu/samd21
[... 147 lines cut ...]
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1209:7d00
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
Determining device status: state = appIDLE, status = 0
Device really in Runtime Mode, send DFU detach request...
dfu-util: error detaching
Device will detach and reattach...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 64
Copying data from PC to DFU device
Download        [=========================] 100%        20236 bytes
Download done.
dfu-util: unable to read DFU status after completion
```
And it results in an uninitialized application. Maybe the error is that I am downloading more than 2 times?

dylad, 2022-08-23:
Alright so I retried on another computer with dfu-util v0.11 and RIOT master branch.

I do get the 
`dfu-util: unable to read DFU status after completion` which is indeed weird. But the application is flashed and start automatically every single time. My board doesn't stay in bootloader after flashing.

Do you get the same behavior than me using RIOT master branch only ? (without your commits).
I'll have a look at this error message to provide a fix soon.

Note: adding dfu-util as PROGRAMMER_SUPPORTED isn't mandatory for quick testing, it will just print a warning but it will use it anyways.

eduazocar, 2022-08-23:
Sorry i tried run again this test in Debian 11 and it works. The bug is present in Ubuntu 22.04 LTS

dylad, 2022-08-23:
There is something weird with ztimer here.
The first time it works fine but all attempts afterwards fail with `dfu-util: unable to read DFU status after completion` becadfu-util: unable to read DFU status after completionuse ztimer doesn't behave properly. The scheduled reboot of 1s is run instantly preventing the device to answer the host.
I'll try to provide a fix in the coming days for this.

dylad, 2022-08-27:
@eduazocar Please have a look at #18520 
This PR should fixes your current issue.

eduazocar, 2022-08-27:
I tested those new changes in #18520, riotboot is working perfectly in Ubuntu 22.04 and on my samr21-xpro board. Thank you very much @dylad

PR #18538 (2022-08-31): bootloader/riotboot_dfu: move ztimer_init() prior to usbus start
### Contribution description

When updating an application to the second `riotboot` slot, `ztimer_init()` is called too late leading to issue with `dfu-util` at the end of the flash process such as:
`dfu-util: unable to read DFU status after completion (LIBUSB_ERROR_PIPE)`

This PR fixes this issue by initializing ztimer prior to setup DFU.

Note: I'm considering dropping ztimer utilisation in the future for USB DFU. But let's fix this issue properly for now.

### Testing procedure
Flash bootloader/riotboot_dfu on your board:
`ma [... cut]
closed 2022-08-31 by maribu