RIOT-OS/RIOT #18458: sam0_sdhc: SDXC cards are not detected / handled properly

benpicco, 2022-08-16:
#### Description

A 64 GB SDXC card shows issues with the `sam0_sdhc` driver.
It works fine with `sdcard_spi`. 

The root cause is that `sam0_sdhc` and `sdcard_spi` both implement the bigh level and low level SD card interface on their own.

It appears that the high level interface of `sdcard_spi` is more complete whereas `sam0_sdhc` fails to detect the SDXH card (it doesn't even enable block addressing).

Ideally the high level code could be shared across drivers.

#### Steps to reproduce the issue

Insert a SD card > 32 GB. Try `tests/mtd_raw` or formatting the file system.

#### Expected results

`test <id>` works with `tests/mtd_raw`, format creates a valid file system.
#### Actual results

Test and format both get stuck.

#### Versions

benpicco, 2022-08-16:
OK, this is weird. I didn't change anything, now it just works

```
2022-08-16 21:40:59,478 # Manual MTD test
2022-08-16 21:40:59,481 # init MTD_0… OK (32 MiB)
2022-08-16 21:40:59,485 # init MTD_1… OK (256 byte)
2022-08-16 21:40:59,679 # init MTD_2… OK (60906 MiB)
> test 2
2022-08-16 21:41:15,499 # [START]
2022-08-16 21:41:15,521 # [SUCCESS]
```

maribu, 2023-01-05:
> ... now it just works

Maybe bad wiring? Let's close this for now, until we have indication that there indeed is an issue.

Or maybe the issue also got fixed as side-affect of fixing the driver on 8-bit platforms?

closed 2023-01-05 by maribu

benpicco, 2023-01-05:
> Or maybe the issue also got fixed as side-affect of fixing the driver on 8-bit platforms?

This is about the native SD host controller driver for the sam0 platform, not the generic SPI one.

maribu, 2023-01-05:
Oh yes, indeed. Sorry for the confusion :sweat_smile: