RIOT-OS/RIOT #18461: LVGL SDL Disp Resizing

tvanfossen, 2022-08-17:
When running a native application using the SDL driver, lvgl_init sets the display size used by the LVGL display driver.

When using an actual LCD, ths display driver ver/hor resolution are set after the driver has been initialized as below:
[image]

For a native application using SDL, I have not been able to resize the display outside of performing a similar action as above, but for SDL.

For reference, I have attempted editing lv_drv_conf.h under the lv_drivers pkg 

[image]

editing SDL_HOR_RES and SDL_VER_RES does not adjust the screen size and the SDL display always runs at the default 320x240 size as defined in lv_disp_drv_init below:

[image]

Alternatively, I also attempted editing lvgl_riot_conf to a similar end, but the result remained an lvgl display window of 320x240.

The SDL window itself will map to the SDL_HOR_RES/SDL_VER_RES size, but the LVGL display window always remains unchanged at 320x240.

In order to resolve this, the simplest solution seemed to be passing LCD_SCREEN_WIDTH/LCD_SCREEN_HEIGHT into the display driver, LCD_SCREEN_WIDTH/HEIGHT being passed in as CFLAGS in a pkg Makefile.include

[image]

Passing disp_dev_width/height into the disp_drv for SDL results in a kernel panic

tvanfossen, 2022-08-17:
Clarification: SDL window and display window are two separate pieces - the SDL window is adjustable in its current form by editing lv_drv_conf or passing a CFLAG with SDL_HOR/VER_RES, but the display driver used by LVGL is NOT editable with the current conf files, when using SDL drivers in a RIOT native application

maribu, 2022-09-14:
I only skimmed this issue, but I think with https://github.com/RIOT-OS/RIOT/pull/18463 merged, this issue can be closed. Please re-open, if this isn't fixed by https://github.com/RIOT-OS/RIOT/pull/18463

closed 2022-09-14 by maribu