RIOT-OS/RIOT #18468: [puf_sram] Hardfault

Carton32, 2022-08-18:
Hi,

I'm using a custom board with a STM32F302RB and a MRF24J40MD. To test it, I've tried to use the example called `gnrc_networking`. By adding `USEMODULE += mrf24j40md` to the Makefile, I can use it without any issue. I'm on RIOT-OS 2021.10.

Now, I've try to use the `puf_sram` module by adding:
```
USEMODULE += puf_sram
USEMODULE += random
```
to the Makefile and:
```
  uint32_t rand = random_uint32();
  printf("RANDOM = %lu\n", rand);
```
to the `main.c` (and include "random.h") after the first puts().

I'm getting the following output:
```
2022-08-18 09:20:45,045 # 
2022-08-18 09:20:45,047 # Context before hardfault:
```

By changing the GCC optimization flag from -Os (default) to -O1 in the Makefile of the example, the application can run perfectly.

Does anyone know what's the cause of this ? I'm asking because on one of my applications which needs `puf_sram`, I'm running out of ROM and it would be nice to be able to use the default optimization to solve my issue.

I'm running an outdated version of RIOT-OS but there wasn't any changes made to the .c file of puf_sram since 2020, same for cpu_conf.h.

Is it a bug or am I doing something wrong ?

PeterKietzmann, 2022-08-18:
- Please put the failing code to a branch, I could try to reproduce this with a nucleo-f303r8 which comes close to your platform (?!).
- Have you tried to compile with docker?

Carton32, 2022-08-22:
Hi,

thanks for your reply. I could find a Nucleo-F303RE and try to reproduce my issue without "success".

I think I can close the issue as it seems to be related with my own code.

About docker, I'm compiling with my own Dockerfile but only for the production releases.

closed 2022-08-22 by Carton32