RIOT-OS/RIOT #18475: Cannot find functions for connection to WiFi network [ESP8266]

luverolla, 2022-08-18:
#### Description
I don't actually know if this can be defined as a feature request. But I looked in the documentation and in the examples folder, and I can't find methods or structures for connections to WiFi networks (init, authentication via SSID and password). I'm using a NodeMCU ESP8266 module, so the WiFi is the only way I have to make my device connected

### Useful links
The methods I'm talking about should be similar to 
[these](https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-reference/wifi/esp_wifi.html)

benpicco, 2022-08-24:
Connecting to a WiFi network works by setting

```
CFLAGS += -DESP_WIFI_SSID=\"your SSID\"
CFLAGS += -DESP_WIFI_PASS=\"your password\"
```

and enabling the `esp_wifi` module.

Currently there is no API to switch the WiFi network at run-time.

maribu, 2023-01-05:
> Does this mean if we have an arduino code connecting to esp8266, and have flashed it, although the CFLAGS parameters are not defined, but in the code we have the ssid and password, we cannot connect to the wifi. Am I correct?

I read this as "Is it correct that we currently cannot change the WiFi connect to at runtime?". Yes, sadly this is not possible yet. There is another issue that requests the feature to scan for WiFi networks and connect to them dynamically at runtime: https://github.com/RIOT-OS/RIOT/issues/18569

Since https://github.com/RIOT-OS/RIOT/issues/18569 is more active, I close this one in favor of https://github.com/RIOT-OS/RIOT/issues/18569

closed 2023-01-05 by maribu