RIOT-OS/RIOT #18493: Bangle.JS2 board support

nica-f, 2022-08-21:
#### Description
This is just a note that I am working on RIOT support for the Bangle.JS2 smart watch (nRF52840 platform).
It already works, with display support, LVGL, touch screen support, some sensors - matching pull requests coming soon :-)

### Useful links
My RIOT fork for board support, LCD driver etc.:
https://github.com/nica-f/RIOT-Digitaluhr
A little test hack app using the above board support:
https://github.com/nica-f/RIOT-Digitaluhr-App

chrysn, 2022-08-21:
I've just tried the fork, just be sure to check out the bangle-js2 branch in the RIOT-Digitaluhr repositiory. Flashing it is really just as simple as connecting the debugger and running `make flash` in the -App repository.

If you, for reasons I can not fathom, should deign to the heresy of reverting from RIOT, I've taken an image of the original firmware ([original.zip](https://github.com/RIOT-OS/RIOT/files/9389332/original.zip) -- don't ask me about the precise license, but it's a conglomerate of the [upstream firmware](https://www.espruino.com/Bangle.js2) and [some version of the softdevice](https://www.nordicsemi.com/Products/Development-software/s140/download)). If you're playing around with RIOT-Digitaluhr-App it's easiest to flash that using:

```
$ cd RIOT-Digitaluhr-Appl/wathcy
$ make debug     # it's just a simple way of starting OpenOCD right
[...]
(gdb) monitor flash write_image erase /tmp/original.bin
(gdb) monitor reset halt
(gdb) c
```

The original.bin file was obtained from one of the watches using the same procedure but `(gdb) monitor flash read_bank 0 /tmp/original.bin` instead.

maribu, 2022-08-30:
With https://github.com/RIOT-OS/RIOT/pull/18478 and https://github.com/RIOT-OS/RIOT/pull/18526 being merged and https://github.com/RIOT-OS/RIOT/pull/18528 on the way in, is anything infrastructure-wise still needed for upstreaming board support?

maribu, 2022-08-30:
Note: I wouldn't mind an incomplete state being PRed. E.g. sensors that are not yet upstream could just be documented as currently unsupported and be added later on. In fact, I think this would streamline the upstreaming process :)

nica-f, 2022-09-03:
I need to really work on this... at least the board support should be mergable, the LCD driver contains some cruft from testing things back and forth I need to clean up.
It's definitely on my todo list!

closed 2026-10-16 by crasbe