RIOT-OS/RIOT #18522: Builds fail when different execstack options are around in objects

chrysn, 2022-08-28:
Something in the latest Debian sid updates (possibly some experimental packages, couldn't pin it down yet) makes builds fail if one of the objects involved has the .note.GNU-stack section present indicating a non-executable stack, and one of the objects has not. This is the case, typically, when Rust programs are built, as the Rust compiler apparently never needs an executable stack.

That'd generally not fatal (it just triggers a "ld: warning: lib_a-nano-msizer.o: missing .note.GNU-stack section implies executable stack" / "ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker"), but as we build with `-Wl,--fatal-warnings`, it becomes fatal as "collect2: error: ld returned 1 exit status".

The best workaround I've found is to explicitly set `-z execstack` or `-z noexecstack` (I'd need to look into the platforms to see what is actually the case, probably execstack, but if we don't use it and have an MPU, maybe make it noexec?); weirdly, both are accepted by the linker.

If this is becoming common even on systems w/o experimental, this may need an urgent fix and backport. (But if so, it'd be a simple-to-test one as it wouldn't affect built binaries, just make the linker happy).

chrysn, 2022-09-20:
I'd consider removing the `-Wl,--fatal-warnings` as this really *is* a warning, and maybe linker authors have gotten wiser in what they classify as critical. (IMO a warning should be just that, an error should be an error).

The flag was introduced in ba2d4a0d116fc082e2b93af06ec57deb3e87f99a as part of https://github.com/RIOT-OS/RIOT/pull/3120 -- @jnohlgard, where would you place this warning in the area between "there is a warning we should address but it's not broken yet" (but `-Wl,--fatal-warnings` makes it break things) and "the binary is probably broken anyway"?

OlegHahm, 2022-10-19:
On OpenSUSE with `GNU ld (GNU Binutils; openSUSE Tumbleweed) 2.39.0.20220810-1` on recent master I get
```
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: warning: /home/oleg/RIOT/examples/hello-world/bin/native/cpu/tramp.o: missing .note.GNU-stack section implies executable stack
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
/usr/lib64/gcc/x86_64-suse-linux/12/../../../../x86_64-suse-linux/bin/ld: warning: /home/oleg/RIOT/examples/hello-world/bin/native/hello-world.elf has a LOAD segment with RWX permissions
```
when building the helloworld example for native.

The last warning also appears on Arch Linux with `GNU ld (GNU Binutils) 2.39.0`.

chrysn, 2022-10-19:
>when building the helloworld example for native.
and anything for cortex boards?

OlegHahm, 2022-10-19:
Nope - neither on Arch nor on OpenSUSE. (Talking about the C helloworld example, not RUST.)

chrysn, 2022-10-19:
Thanks, and just to check: you tested with TOOLCHAIN=llvm?

OlegHahm, 2022-10-19:
Nope, I tested with `TOOLCHAIN` set to the default value (gcc I assume). With `TOOLCHAIN=llvm` I don't any of these warnings.