RIOT-OS/RIOT #18534: `tests/thread_priority_inversion` failing with `esp32-wroom-32

MrKevinWeiss, 2022-08-30:
#### Description

Nightlies and any PR requiring `CI: run tests` are failing due to this.

#### Steps to reproduce the issue

```
BOARD=esp32-wroom-32 make flash test -C tests/thread_priority_inversion
```

Or look at the nightlies

#### Expected results
Test passes

#### Actual results
Test fails
<details>

```
-- running on worker pi-67e8efb7 thread 1, build number 24423.
-- executing tests for tests/thread_priority_inversion on esp32-wroom-32 (compiled with gnu toolchain):
- executing hook run_test_pre
- resetting USB power...
- waiting for tty to re-attach...
- hook run_test_pre finished
make: Entering directory '/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion'
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 --before default_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 /tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion/bin/esp32-wroom-32/esp_bootloader/bootloader.bin 0x8000 /tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion/bin/esp32-wroom-32/partitions.bin 0x10000 /tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion/bin/esp32-wroom-32/tests_thread_priority_inversion.elf.bin
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 30:ae:a4:f4:7a:e0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0320
Compressed 17088 bytes to 11709...

Writing at 0x00001000... (100 %)
Wrote 17088 bytes (11709 compressed) at 0x00001000 in 0.3 seconds (effective 427.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 85...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (85 compressed) at 0x00008000 in 0.0 seconds (effective 3649.1 kbit/s)...
Hash of data verified.
Compressed 103952 bytes to 45598...

Writing at 0x00010000... (33 %)
Writing at 0x00014000... (66 %)
Writing at 0x00018000... (100 %)
Wrote 103952 bytes (45598 compressed) at 0x00010000 in 1.2 seconds (effective 667.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
make: Nothing to be done for 'termdeps'.
make: Leaving directory '/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion'
make: Entering directory '/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion'
r
socat - open:/dev/ttyUSB0,b115200,echo=0,raw,cs8,parenb=0,cstopb=0 
r
r
r
r
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DOUT, clock div:2

load:0x3fff0030,len:1360

load:0x40078000,len:12788

ho 0 tail 12 room 4

load:0x40080400,len:2848

entry 0x40080410

Pro cpu up.

Single core mode

Traceback (most recent call last):
  File "/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion/tests/01-run.py", line 21, in <module>
    sys.exit(run(testfunc))
  File "/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/dist/pythonlibs/testrunner/__init__.py", line 28, in run
    logfile=sys.stdout if echo else None)
  File "/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/dist/pythonlibs/testrunner/spawn.py", line 95, in setup_child
    sync_child(child, env)
  File "/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/dist/pythonlibs/testrunner/spawn.py", line 134, in sync_child
    TEST_INTERACTIVE_DELAY)
  File "/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/dist/pythonlibs/testrunner/utils.py", line 30, in test_utils_interactive_sync
    _test_utils_interactive_sync(child, retries, delay)
  File "/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/dist/pythonlibs/testrunner/utils.py", line 22, in _test_utils_interactive_sync
    child.expect_exact(ready_exp, timeout=0)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/spawnbase.py", line 421, in expect_exact
    return exp.expect_loop(timeout)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/expect.py", line 181, in expect_loop
    return self.timeout(e)
  File "/usr/local/lib/python3.5/dist-packages/pexpect/expect.py", line 144, in timeout
    raise exc
pexpect.exceptions.TIMEOUT: Timeout exceeded.
<pexpect.pty_spawn.spawn object at 0x769d2ad0>
command: /usr/bin/make
args: [b'/usr/bin/make', b'cleanterm']
buffer (last 100 chars): 'de\r\r\n'
before (last 100 chars): 'o 0 tail 12 room 4\r\r\nload:0x40080400,len:2848\r\r\nentry 0x40080410\r\r\nPro cpu up.\r\r\nSingle core mode\r\r\n'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 3554
child_fd: 5
closed: False
timeout: 10
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_string:
    0: 'READY'
/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/makefiles/tests/tests.inc.mk:22: recipe for target 'test' failed
make: *** [test] Error 1
make: Leaving directory '/tmp/dwq.0.356379192510457/3af8f868589d03099f720231dddc28e8/tests/thread_priority_inversion'
```
</details>

#### Versions
Fails on the CI

gschorcht, 2022-08-31:
The problem also occurs for ESP32-C3, ESP32-S2, ESP32-S3.

gschorcht, 2022-08-31:
OK, the test crashes on ESP32 SoCs here: https://github.com/RIOT-OS/RIOT/blob/0e2eb919a4f437ebe8c3141a8c4059ac0e0bc8dc/core/mutex.c#L94-L96 Reason is that the threadsafe `malloc` function which uses a mutex https://github.com/RIOT-OS/RIOT/blob/0e2eb919a4f437ebe8c3141a8c4059ac0e0bc8dc/sys/malloc_thread_safe/malloc_wrappers.c#L29-L35 is already called during the system initialization before the threads are created so that `thread_get_active` returns 0.

I guess that we have to fix the threadsafe `malloc` function so that the mutex is only used when threads are already created and scheduling is started.

PR #18544 (2022-09-05): cpu/esp: improve thread safety in newlib locking functions
### Contribution description

This PR improves the thread safety of the newlib locking functions as following:

- Dynamic allocation and initialization of the mutex used by a newlib locking variable must not be interrupted. Since a thread context switch can occur on exit from an ISR, the allocation and initialization of such a mutex is guarded by disabling interrupts. The same is done for the release of such a mutex.
- Locking functions implicitly allocate and initialize the mutex used by a newlib locking variable if necessary. This im [... cut]
closed 2022-09-05 by gschorcht