RIOT-OS/RIOT #18552: Support for podman as alternative to docker

keestux, 2022-09-04:
#### Description
Since I discovered podman [1] I have said goodbye to docker. podman is command line compatible with docker, so I can almost build RIOT (using `BUILD_IN_DOCKER=1`).

I don't want to shove podman in everyone's throat, but I hope we can add podman support in the RIOT build system. The following diff is basically all that is needed to build with podman.
```
diff --git a/makefiles/docker.inc.mk b/makefiles/docker.inc.mk
index 79f1d292e1..60836e6df4 100644
--- a/makefiles/docker.inc.mk
+++ b/makefiles/docker.inc.mk
@@ -1,4 +1,4 @@
-export DOCKER_IMAGE ?= riot/riotbuild:latest
+export DOCKER_IMAGE ?= docker.io/riot/riotbuild:latest
 export DOCKER_BUILD_ROOT ?= /data/riotbuild
 DOCKER_RIOTBASE ?= $(DOCKER_BUILD_ROOT)/riotbase
 # List of Docker-enabled make goals
@@ -123,8 +123,7 @@ DOCKER ?= docker
 # - allocate a pseudo-tty
 # - remove container on exit
 # - set username/UID to executor
-DOCKER_USER ?= $$(id -u)
-DOCKER_RUN_FLAGS ?= --rm --tty --user $(DOCKER_USER)
+DOCKER_RUN_FLAGS ?= --rm --tty --userns keep-id
 
 # allow setting make args from command line like '-j'
 DOCKER_MAKE_ARGS ?=
```

Two changes
1) podman needs to be told explicitly that the registry is at `docker.io`. I don't know if `docker` understands that syntax too
2) instead of `--user <uid>` I would like to user podman's `--userns keep-id`

That should be doable. All it needs is some Makefile trickery :-)

### Useful links
[1] https://podman.io

keestux, 2022-09-04:
To confirm for change 1. Yes, `docker` is equally happy with this syntax: `docker.io/riot/riotbuild:latest`

maribu, 2022-09-14:
I personally see an advantage in allowing user to use `podman` instead of docker. Feel free to open a PR. I am happy to test with both `podman` and `docker` to make sure nobody is left behind.

PR #18671 (2022-10-05): Add podman support
### Contribution description

This change allows the use of podman instead of docker.

### Testing procedure

On a system where `podman` is installed, build an example
```
BUILD_IN_DOCKER=1 make -C examples/hello-world DOCKER=podman BOARD=sodaq-sara-sff clean all
```
If the docker command is redirected to podman you don't need the `DOCKER=podman`.
```
$ BUILD_IN_DOCKER=1 make -C examples/hello-world BOARD=sodaq-sara-sff clean all
```

On a system where `docker` is installed do:
```
$ BUILD_IN_DOCKER=1 make -C examples/hello-world BOARD=sodaq-sara-sff clean all
```

### Issue [... cut]
closed 2022-10-05 by benpicco