RIOT-OS/RIOT #18566: Error with make for ESP32 blinky example

SrivamsiMalladi, 2022-09-06:
#### Expected results

#### Actual results

#### Versions

**Description**
Getting error running make fro ESP32 with the blinky example. Installed ESP-IDF properly. Ran esp-idf/export.sh in the current directly in the terminal.

**Steps to reproduce the issue**
* Build blinky example from RIOT-OS/RIOT/examples

**Expected results**
Successful make run

**Actual results**
iotdev@iotdev:~/RIOT/examples/blinky$ make all
Building application "blinky" for "esp32-wroom-32" with MCU "esp32".

"make" -C /home/iotdev/riot-os/RIOT/pkg/esp32_sdk/ 
"make" -C /home/iotdev/riot-os/RIOT/boards/common/init
"make" -C /home/iotdev/riot-os/RIOT/boards/esp32-wroom-32
"make" -C /home/iotdev/riot-os/RIOT/boards/common/esp32
"make" -C /home/iotdev/riot-os/RIOT/core
"make" -C /home/iotdev/riot-os/RIOT/core/lib
"make" -C /home/iotdev/riot-os/RIOT/cpu/esp32
/home/iotdev/riot-os/RIOT/cpu/esp32/startup.c: In function 'system_init':
/home/iotdev/riot-os/RIOT/cpu/esp32/startup.c:243:31: error: implicit conversion from 'enum <anonymous>' to 'esp_log_level_t' [-Werror=enum-conversion]
  243 |     esp_log_level_set("wifi", LOG_DEBUG);
      |                               ^~~~~~~~~
/home/iotdev/riot-os/RIOT/cpu/esp32/startup.c:244:31: error: implicit conversion from 'enum <anonymous>' to 'esp_log_level_t' [-Werror=enum-conversion]
  244 |     esp_log_level_set("gpio", LOG_DEBUG);
      |                               ^~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [/home/iotdev/riot-os/RIOT/Makefile.base:146: /home/iotdev/riot-os/RIOT/examples/blinky/bin/esp32-wroom-32/cpu/startup.o] Error 1
make[1]: *** [/home/iotdev/riot-os/RIOT/Makefile.base:31: ALL--/home/iotdev/riot-os/RIOT/cpu/esp32] Error 2
make: *** [/home/iotdev/riot-os/RIOT/Makefile.include:738: application_blinky.module] Error 2

SrivamsiMalladi, 2022-09-06:
Running `. ../../dist/tools/esptools/export.sh esp32` from https://doc.riot-os.org/group__cpu__esp32.html#autotoc_md40 worked, not the `esp-idf/export.sh` as suggested in the ESP-IDF documentation.

benpicco, 2022-09-07:
Please refer to the RIOT documentation on how to set up the toolchain for RIOT.
We have some patches on top of the ESP-IDF to make it compatible with RIOT, so you can't use the upstream SDK directly.

There are scripts in `dist/tools/esptools` that will take care of downloading & patching the correct toolchain.

closed 2022-09-07 by benpicco