RIOT-OS/RIOT #18569: Missing Wi-Fi features for ESP32

SrivamsiMalladi, 2022-09-08:
#### Description

I am looking forward to using RIOT-OS with ESP32 for developing IoT applications for which I will need certain features like
* Switching between Station mode and SoftAP mode during runtime,
* Scanning available networks during runtime (like the [esp_wifi_scan_start()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_scan_startPK18wifi_scan_config_tb) from the ESP-IDF's Wi-Fi Driver)
* Connecting to a specific Wi-Fi network during runtime i.e. specifying the SSID and password of the network to connect to during runtime

From my understanding of the RIOT-OS's WiFi Network Interface for ESP32, it is
* only possible to specify the mode (Station or SoftAP) in the makefile before building the application
* not possible to scan for available networks
* only possible to specify the SSID and password of the network to connect to in the makefile or during the make command

Please let me know if these features are already available or if I am missing something related to these features.

### Useful links

* [RIOT-OS's Wi-Fi Network Interface for ESP32](https://doc.riot-os.org/group__cpu__esp32.html#esp32_wifi_network_interface)
* [ESP-IDF's Wi-Fi Driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html)

gschorcht, 2022-09-12:
> From my understanding of the RIOT-OS's WiFi Network Interface for ESP32, it is
>
> * only possible to specify the mode (Station or SoftAP) in the makefile before building the application
> * not possible [...]

You are absolutely right, the WiFi interface is statically configured at compile time. There have been no plans to change this yet.

The reason for this is the design of how RIOT handles network devices. In RIOT, network devices are statically configured at compile time and enabled at system startup. For the protocol stack, the network device is then just available regardless of whether it is an Ethernet or WiFi interface. Thus, there was no need to change the WiFi interface mode, change WiFi parameters at runtime, or search for available WiFi networks.

The correct approach would to implement such features would be
- to add the functions which implement such features to the [`esp_wifi` netdev driver](https://github.com/RIOT-OS/RIOT/tree/master/cpu/esp_common/esp-wifi),
- to add additional [netopt configuration options](https://doc.riot-os.org/group__net__netopt.html) like `NETOPT_WIFI_SSID`, `NETOPT_WIFI_AP_AVAIL`,  a.s.o., and
- to use the [Network Device Driver API](https://doc.riot-os.org/group__drivers__netdev__api.html) functions `get` and `set` to call these functions.

You are welcome to contribute your ideas or even better the code :smile:

SrivamsiMalladi, 2022-09-21:
Hello @gschorcht. I would like to try implementing these features. I would like to start with adding support for changing the SSID and PASS in STA mode during runtime. So, if I understood correctly, I need to 
* add `NETOPT_WIFI_SSID`, `NETDEV_WIFI_PASS` to the [enum netopt_t](https://doc.riot-os.org/group__net__netopt.html#ga19e30424c1ab107c9c84dc0cb29d9906) in the [netopt.h (netopt configuration options](https://doc.riot-os.org/group__net__netopt.html))
* In the [esp_wifi](https://github.com/RIOT-OS/RIOT/tree/master/cpu/esp_common/esp-wifi), I need to add the corresponding cases for these new netopts in the switch statements of [`_esp_wifi_get()`](https://github.com/RIOT-OS/RIOT/blob/master/cpu/esp_common/esp-wifi/esp_wifi_netdev.c#L689) and [`_esp_wifi_set()`](https://github.com/RIOT-OS/RIOT/blob/master/cpu/esp_common/esp-wifi/esp_wifi_netdev.c#L730) fucntions.

Then to use the functionality I wanted, I need to set the SSID and PASS in the netdev (esp_wifi_netdev_t) using the netdev.set(), which is currently being initialized by the auto_init_esp_wifi() in the init.c. My question is, when I change the SSID and PASS using .set(), does the ESP32 automatically try to connect to this new Wi-Fi, or some reload()/reinit() kind of function must be called to reinitialize the Wi-Fi driver to consider the new Wi-Fi credentials?

> ( I have only used drivers/internal files directly till now. Never tried to mess around with the internal driver files. So. sorry if I am asking too many or silly questions.)