RIOT-OS/RIOT #18576: nRF53 support

chrysn, 2022-09-11:
#### Description

Nordic has released a new series of microcontrollers with radio modules, nRF53, with its flagship nrf5340

It would be great to have support for them; relevant sub-items are:

* Support for the new radios. Most urgent is probably Bluetooth (as they offer new BLE features like isochronous transfers).

  As nimble already [supports them](https://github.com/apache/mynewt-nimble/pull/963), might just be a nimble update and some glue code away?

* Multiprocessor support.

  I don't think we have to go all in immediately, even though asymmetric multiprocessing should be a long-term goal. Being able to run on the radio core alone should already get us some of the way.

#### Relevant boards

Boards that should be easy to add once the family is in are the thingy:53, nrf5340-dk and nrf5340-audio-dk.

(Generally, BLE audio is a really big new item in these, and that will be in high demand, judging from feedback from a hearing aid technician).

chrysn, 2022-09-19:
I've had a look at the nRF5340 documentation and stay by my initial idea that the network core is a good initial target. It doesn't have all the peripherals the application core has (it only has one WDT while the application core has two, one UART where the application has 4, has no access to the PDM microphone driver and most importantly no USB) -- but it has access to RADIO, and all memory (flash and RAM) of the application core is also available to the MCU core (but with no DMA, and application flash is not executable).

By the way, the RADIO peripheral appears to be well documented, I haven't seen any hint at the necessity to use the softdevice other for than for convenience and easier certification.

sjanc, 2023-02-02:
Just my 2 cents :)

Nimble (on Mynewt) on nRF5340 uses "classic split" architecture: application core runs host, networking core runs controller only and both communicates with standard HCI (via IPC transport).

If you start with support for application core first you may use Mynewt based blehci on networking core (if you reuse IPC HCI transport) and only later move to RIOT based controller (if needed).

I'd suggest starting with app core support first for one more reason:  networking core is by default powered down and its reset needs to be pulled down from app core so you need "something" working on app core anyway to play with networking core (this is how I added support for nRF5340 in Mynewt, first app core, then networking core).

And last but not least, in latest NimBLE master nRF52x and nRF5340 PHYs are combined into single nRF5X driver, so updating NimBLE may also ease you with enabling radio on nrf53 if you already support nrf52 :)

dylad, 2023-06-26:
I've started to work on adding nrf5340dk board to RIOT-OS.
I'll focus on basic stuff at first (common nRFx peripherals).

dylad, 2023-07-05:
Initial support for the application core was merged. Should we keep this one open ? Or should we migrate further discussions in #19267 ?

crasbe, 2026-10-06:
No need to keep this open, given the nRF tracking issue.

closed 2026-10-06 by crasbe