RIOT-OS/RIOT #18622: Command line CoAP client does not support queries

chrysn, 2022-09-21:
#### Description

The command line CoAP client of the gcoap example does not support sending query parameters. This is in line with the documentation (although not with local variable names), as docs say the syntax is `coap <get|post|put> [-c] <addr>[%iface] <port> <path> [data]` (i.e. path, not path-and-query).

Users who see that this works with `/foo/bar` might be believe that `/foo/bar?qux=1&quux=2` also works -- but it doesn't, it turns into `/foo/bar%3Fqux=1&quux=2` (which has two path segments, "foo" and "bar?qux=1&qux=2" rather than having a "foo" and a "bar" path segment and a "qux=1" and a "quux=2" query segment).

(Inconsistently, when using a proxy -- then, the question mark is used properly, due to a URI being assembled in a Proxy-Uri option).

Down the call stack, uses `gcoap_req_init`, which also takes a path argument `path`.

Here we have two paths forward:

* Declare that every time we have a convenience "path" argument, we really mean "path and query", and split it up inside `gcoap_req_init` and its callees.
* Insist that "path" means "path" and not "local URI part". If we extend the CLI example to also process queries, it'd either ask users to provide path and query in separate space seperated arguments (super weird, but then again we already take host and port the same way) or find the query on our own, go through [gcoap_req_init_path_buffer](https://doc.riot-os.org/group__net__gcoap.html#ga034e3fe03d8398845e317a29c72d9605) and then add the query option separately.

I have no strong preference -- who has opinions? @haukepetersen, @benpicco, @Teufelchen1 or @miri64?

If left to my own devices and unguided, I'd probably leave this until #13827 takes over the URI world hahaha 😈

### Issue references

Split out from https://github.com/RIOT-OS/RIOT/issues/17986

miri64, 2022-09-22:
I'd prefer if the command line client just supports URIs instead of a set of parameters. This way, the client would also support both CoAP and CoAPS finally.

chrysn, 2022-09-22:
Conveniently, @teuelchen1 already did that for CoRE-RD in #18053, that
could be the protoype template (and provide helpers already)