RIOT-OS/RIOT #18662: SAMR30/I2C: How to send I2C address + W bit not followed by data?

biboc, 2022-09-28:
I'm doing a driver for a chip. To wake it up, I need to send its [I2C address] + Write bit + ACK + STOP on the I2C bus ONLY, no data needed to be written after
Problem is that I can't do it with current sam0 I2C implementation because it stops the write function if I don't give data
Should we remove this condition?
https://github.com/RIOT-OS/RIOT/blob/09fd98c0c6f13c9dba2cbce7c4301465f4bfdab9/cpu/sam0_common/periph/i2c.c#L287

dylad, 2022-09-30:
> To wake it up, I need to send its [I2C address] + Write bit + ACK + STOP on the I2C bus ONLY, no data needed to be written after

Do you mind share with us the device reference ?  It would be interesting to have a look at the datasheet.

Can we simply send a dummy byte after ? Or does it mess up the chip ?

If we cannot send  a dummy byte, my suggestion do deal with such kind of device would be to introduce a [new I2C flag](https://github.com/RIOT-OS/RIOT/blob/d0ac322a117f22f338cd88b8cf40306ec1bbbc71/drivers/include/periph/i2c.h#L195) to our API to skip the condition you mentioned only if this flag is present. But I am wondering if all I2C MCU IP would support it...

biboc, 2023-04-27:
I finally tested with a dummy byte and it works, the chip (HS300X) ignored the byte
Apart if someone else needs to send I2C data without a bye, I close this issue

dylad, 2023-04-27:
Great ! Thanks for the update.