RIOT-OS/RIOT #18748: Can't build rust hello world example.

omicronns, 2022-10-14:
Hi,
I'm trying to build `rust-hello-world` example and it fails at linking step:

```
➜  rust-hello-world git:(2022.10-branch) make BOARD=cc1352-launchpad          
Building application "hello-world" for "cc1352-launchpad" with MCU "cc26x2_cc13x2".

"make" -C /home/kad/Documents/Work/projects/RIOT/boards/cc1352-launchpad compile-commands
make[2]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/boards/common/init compile-commands
make[2]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/core compile-commands
make[2]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/core/lib compile-commands
make[2]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26x2_cc13x2 compile-commands
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26x2_cc13x2/periph compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26xx_cc13xx compile-commands
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26xx_cc13xx/periph compile-commands
make[4]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cortexm_common compile-commands
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cortexm_common/periph compile-commands
make[4]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/drivers compile-commands
"make" -C /home/kad/Documents/Work/projects/RIOT/drivers/periph_common compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/sys compile-commands
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/auto_init compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/div compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/libc compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/malloc_thread_safe compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/newlib_syscalls_default compile-commands
make[3]: Nothing to be done for 'compile-commands'.
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/stdio_uart compile-commands
make[3]: Nothing to be done for 'compile-commands'.
    Finished release [optimized] target(s) in 0.04s
"make" -C /home/kad/Documents/Work/projects/RIOT/boards/cc1352-launchpad
"make" -C /home/kad/Documents/Work/projects/RIOT/boards/common/init
"make" -C /home/kad/Documents/Work/projects/RIOT/core
"make" -C /home/kad/Documents/Work/projects/RIOT/core/lib
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26x2_cc13x2
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26x2_cc13x2/periph
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26xx_cc13xx
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cc26xx_cc13xx/periph
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cortexm_common
"make" -C /home/kad/Documents/Work/projects/RIOT/cpu/cortexm_common/periph
"make" -C /home/kad/Documents/Work/projects/RIOT/drivers
"make" -C /home/kad/Documents/Work/projects/RIOT/drivers/periph_common
"make" -C /home/kad/Documents/Work/projects/RIOT/sys
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/auto_init
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/div
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/libc
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/malloc_thread_safe
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/newlib_syscalls_default
"make" -C /home/kad/Documents/Work/projects/RIOT/sys/stdio_uart
/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: warning: lib_a-nano-vfprintf_i.o: missing .note.GNU-stack section implies executable stack
/usr/lib/gcc/arm-none-eabi/12.2.0/../../../../arm-none-eabi/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
collect2: error: ld returned 1 exit status
make: *** [/home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/../../Makefile.include:729: /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/hello-world.elf] Error 1
```

Dry run for more information:

```
➜  rust-hello-world git:(2022.10-branch) make -n BOARD=cc1352-launchpad
[... 119 lines cut ...]
# ar to unpack it...
mkdir -p /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/rust_hello_world/bin/
cd /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/rust_hello_world/ && arm-none-eabi-ar x /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/target/thumbv7em-none-eabihf/release/librust_hello_world.a
# ... and move them back if any exist, careful to err if anything is duplicate
rmdir /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/rust_hello_world/bin/ || (mv -n /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/rust_hello_world/bin/* /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/rust_hello_world/ && rmdir /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/rust_hello_world/bin/)
arm-none-eabi-gcc $(find /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/rust_hello_world/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/application_hello-world/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/auto_init/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/board/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/board_common_init/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/cc26xx_cc13xx/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/cc26xx_cc13xx_periph/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/core/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/cortexm_common/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/cortexm_common_periph/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/cpu/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/div/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/libc/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/malloc_thread_safe/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/newlib_syscalls_default/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/periph/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/stdio_uart/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/sys/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/core_lib/ /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/periph_common/ -name "*.o" 2> /dev/null | sort)  -lm -L/home/kad/Documents/Work/projects/RIOT/cpu/cc26x2_cc13x2/ldscripts -L/home/kad/Documents/Work/projects/RIOT/cpu/cortexm_common/ldscripts -T/home/kad/Documents/Work/projects/RIOT/cpu/cc26xx_cc13xx/ldscripts/cc26xx_cc13xx.ld -Wl,--fatal-warnings -mcpu=cortex-m4 -mlittle-endian -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ggdb -g3 -Os -static -lgcc -nostartfiles -Wl,--gc-sections -Wl,--defsym=_rom_start_addr=0x00000000 -Wl,--defsym=_ram_start_addr=0x20000000 -Wl,--defsym=_rom_length=352K -Wl,--defsym=_ram_length=80K   -L/home/kad/Documents/Work/projects/RIOT/core/ldscripts -Wl,-wrap=malloc -Wl,-wrap=calloc -Wl,-wrap=realloc -Wl,-wrap=free -specs=nano.specs -lc -Wl,-Map=/home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/hello-world.map -o /home/kad/Documents/Work/projects/RIOT/examples/rust-hello-world/bin/cc1352-launchpad/hello-world.elf
[... cut ...]
        /usr/bin/env echo -e "RIOT standard peripherals code (1209/7D00) cannot be set explicitly." 1>&2 ; \
        /usr/bin/env echo -e "Unset USB_VID / USB_PID for the code to be picked automatically, or set" 1>&2 ; \
        /usr/bin/env echo -e "them to \${USB_VID_TESTING} / \${USB_PID_TESTING} during development." 1>&2 ; \
        exit 1; \
fi
```
To be able to compile the code I had to remove `-Wl,--fatal-warnings` from `makefiles/arch/cortexm.inc.mk`, which I think is not intended solution.

Is this known problem?

maribu, 2023-01-05:
Sorry for the long latency. Would you mind to paste the output of

```
$ make print-versions
```

? Could you also retry again with the current `master` of RIOT? I think we may have merged a PR that fixes the issue in the meantime without noticing it is relevant to this issue.

maribu, 2023-05-17:
Since the reporter is unresponsive, I'm closing this. Building rust apps in `master` works fine ... almost. Once https://github.com/RIOT-OS/RIOT/pull/19602 is in it should indeed work again.

closed 2023-05-17 by maribu