RIOT-OS/RIOT #18766: potential stack corruption in use of isrpipe

Enoch247, 2022-10-18:
#### Description

I belive the line `CFLAGS += -Wno-cast-function-type` [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/stdio_uart/Makefile#L2) is a bug. It silences a warning thrown [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/stdio_uart/stdio_uart.c#L57). The uart module expects a pointer to a function with no return value, but `isrpipe_write_one()` returns an int. It may not cause an issue for some ABIs, but could lead to stack corruption if the ABI does not use frame pointers.

This error is also silenced in a few other places:

```
$ git grep Wno-cast-function-type
cpu/esp32/esp-idf/ble/Makefile:CFLAGS += -Wno-cast-function-type
pkg/wamr/Makefile:CFLAGS += -Wno-cast-function-type
sys/stdio_uart/Makefile:  CFLAGS += -Wno-cast-function-type
```

#### Steps to reproduce the issue

Remove the compiler flag mentioned, and observe warning.

#### Expected results

N/A

#### Actual results

N/A

#### Versions

```
$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/9.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/lib/include' --mandir='/usr/lib/share/man' --infodir='/usr/lib/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --mandir=/usr/share/man --enable-languages=c,c++,lto --enable-multilib --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --enable-tls --build=x86_64-linux-gnu --target=arm-none-eabi --with-system-zlib --with-gnu-as --with-gnu-ld --with-pkgversion=15:9-2019-q4-0ubuntu1 --without-included-gettext --prefix=/usr/lib --infodir=/usr/share/doc/gcc-arm-none-eabi/info --htmldir=/usr/share/doc/gcc-arm-none-eabi/html --pdfdir=/usr/share/doc/gcc-arm-none-eabi/pdf --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --disable-libstdc++-v3 --host=x86_64-linux-gnu --with-headers=no --without-newlib --with-multilib-list=rmprofile CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' LDFLAGS='-Wl,-Bsymbolic-functions -Wl,-z,relro' OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-arm-none-eabi-Gl9kT9/gcc-arm-none-eabi-9-2019-q4=. -fstack-protector-strong' INHIBIT_LIBC_CFLAGS=-DUSE_TM_CLONE_REGISTRY=0 AR_FOR_TARGET=arm-none-eabi-ar AS_FOR_TARGET=arm-none-eabi-as LD_FOR_TARGET=arm-none-eabi-ld NM_FOR_TARGET=arm-none-eabi-nm OBJDUMP_FOR_TARGET=arm-none-eabi-objdump RANLIB_FOR_TARGET=arm-none-eabi-ranlib READELF_FOR_TARGET=arm-none-eabi-readelf STRIP_FOR_TARGET=arm-none-eabi-strip
Thread model: single
gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (15:9-2019-q4-0ubuntu1)
```

maribu, 2022-10-19:
Yes, indeed. The function signature is casted from `int (*fn)(isrpipe_t *, uint8_t)` to `void (fn)(void *, uint8_t)`. This is undefined behavior.

I think in practice, the calling conventions of currently supported platforms pass `isrpipe_t *` the same way the pass `void *` and pass the return value in a caller safe register. But I am not even sure that this is the case for the currently supported platforms, even more I fear this may not be true for platforms added in the future. And even when this is well defined on a given platform, it is undefined behavior from the C lang point of view.

There have been numerous examples in the past where C compilers generated the intended machine code from the source despite undefined behavior, but stopped doing so after an update. We will certainly be better off to fix the issue.

Thanks for reporting this!

maribu, 2022-10-19:
See https://github.com/RIOT-OS/RIOT/pull/18774 for the stdio_uart one.

Maybe @kfessel could look into WAMR?

With `cpu/esp32` it is relatively easy to reason about whether two function signatures are compatible in regard to the calling convention, as only Xtensa and RISC-V MCUs of a single vendor are used. And since the toolchain is updated there in lockstep with the affected IDF code anyway, I doubt this will bite users when future GCC versions start seeing this undefined behavior as optimization potential. I'd personally rather avoid undefined behavior even at the cost of some extra ROM size, though.

kfessel, 2022-10-20:
It is need for WAMR since it may call into native functions whose signature is not know at compile time of WAMR

kfessel, 2022-10-20:
The data type * to void * cast should be safe on all architectures since void * points to the same addressspace as data * (malloc return void *).
The int vs void return should be safe for all practical implementations since old c implementation had an implicit int return -> void is very likely ABIwise to be handled like int without setting the value in callee and without reading the value in caller. But the fix that @maribu provided solves that.

maribu, 2023-01-05:
The current state is that the issue still is present in the ESP32 IDF and in the WAMR package, correct?

I'm inclined to just say "trust upstream to do the right thing" (tm) and shift the blame towards them, if anything breaks. I think with the calling convention of Xtensa and RISC-V the ESP32 function cast is likely harmless (I think @gschorcht already confirmed it is), at least until a new compiler starts "harvesting the optimization potential" of the cast actually being undefined behavior and optimizes out the function call or something. But since Espressif is closely managing both IDF and toolchain by never upstreaming their toolchain stuff, we can be sure those guys will be the first to notice and fix the IDF when updating their toolchain.

I'm not so sure about WAMR, though.

> The data type * to void * cast should be safe on all architectures since void * points to the same addressspace as data * (malloc return void *).

This is true for all practical architectures I am aware of. Bug the C standard explicitly allows pointer to have a different length depending on the type.

E.g. let's say we have an 8 bit architecture with 18 bit address space. We could have two load instructions in hardware:

- `ldr3 $x $y z`
    - Loads data from address `((($y << 16) | ($(y+1) << 8) | $(y+2)) & 0x3ffff) + z` and stores it in register `$x`
- `ldr2 $x $y z`
    - Loads data from address `(($y << 18) | ($(y+1) << 10)) + z` and stores it in register `$x`

With that architecture, the compiler would most likely use `ldr3` and three byte addresses for `char *` pointers, but `ldr2` and two byte addresses for `uint32_t *`. And the C standard explicitly allows pointers to be of different size. It only mandates that conversion to `void *` and back is loss-less.

kfessel, 2023-01-12:
WAMR supports a specific set of architectures (for its native calls) implementing the calling convention of these architectures in assembly (converting from internal wamr calling convention (a stack) to native calling convention (likely registers and stack)), the function casting for theses architectures works as expected (but needs the warning to be turned off).
There is a fallback implementation in c, that should never be used -and is not by our pkg-, that one doesn't even work for architectures, that are otherwise supported if the wrong optimization level is chosen.

maribu, 2023-01-13:
@kfessel Thx for checking!

My conclusion now is that the remaining instances of function signature casting are closely monitored by the respective upstream devs to not cause runtime issues. Hence, I close this as fixed.

Please reopen, if anyone disagrees.

closed 2023-01-13 by maribu