RIOT-OS/RIOT #18911: setup bors (tracking issue)

kaspar030, 2022-11-15:
This issue is for tracking the setup of bors for our CI.

Non-invasive, preparatory:
- [ ] make ci.riot-os.org build "testing" and "trying"
- [ ] add bors.toml https://github.com/RIOT-OS/RIOT/pull/18910
- [ ] add RIOT-OS/RIOT to bors

Switch over:
- [ ] implement quick build for PR / full build for "testing" in `./.murdock`
- [ ] make bors mandatory for merging (disables manual merges)

Optimizations:
- [ ] allow multiple parallel builds on ci.riot-os.org (so PR builds don't have to wait for "testing" builds)
  - [ ] enable on ci.riot-os.org
  - [ ] make murdock dwq backend prioritize PR jobs

maribu, 2023-01-04:
@kaspar030: Ping?

kaspar030, 2023-01-04:
Migration considered done. I've moved the planned optimizations to another issue: #19092

closed 2023-01-04 by kaspar030