RIOT-OS/RIOT #18957: SAMR30/PWM: Usage of PWM on GPIO_PIN(PA, 9) in mux GPIO_MUX_F with TCC1 on channel 3 does not work

biboc, 2022-11-23:
#### Description
Usage of PWM on GPIO_PIN(PA, 9) in mux GPIO_MUX_F with TCC1 on channel 3 does not work

#### Steps to reproduce the issue
Configure your board with:
```
/**
 * @name    PWM configuration
 * @{
 */
#define PWM_0_EN            1
#define PWM_MAX_CHANNELS    1
/* for compatibility with test application */
#define PWM_0_CHANNELS      PWM_MAX_CHANNELS

/* PWM device configuration */
static const pwm_conf_t pwm_config[] = {
#if PWM_0_EN
    {TCC1, {
        /* GPIO pin, MUX value, TCC channel */
        { GPIO_PIN(PA, 9), GPIO_MUX_F, 3}
    }},
#endif
};
```
And try to change PWM(0) channel 0. 
I'm able to use:
* TCC0 with pin GPIO_PIN(PA,18) MUX F channel 2
* TCC0 with pin GPIO_PIN(PA,19) MUX F channel 3
* TCC2 with pin GPIO_PIN(PA,16) MUX E channel 0
* TCC2 with pin GPIO_PIN(PA,17) MUX E channel 1

I observe now that TCC1 has only 2 Compare/Capture units so it is impossible to use TCC1 with PA 9? 
`#define TCC1_CC_NUM                 2        // Number of Compare/Capture units`

So why the datasheet mentions TCC1 with WO[3]
[image]

There is something I don't understand here
Thanks for your explanation

#### Expected results
The PWM should start and change according to command

#### Actual results
Nothing happens

#### Versions
Ubuntu 18
Old RIOT_VERSION = 2019.01 (I know I need to upgrade :) )

biboc, 2022-11-23:
I tried on master
To reproduce, create a new app with 
```
int main(void) {
  if (pwm_init(PWM_DEV(0), PWM_LEFT, 500, 255) == 0) {
    printf("PWM 0 init PWM fail\n");
    return -1;
  }
  printf("Init PWM 0 OK\n");
 
  while (1) {
    xtimer_sleep(2);
    printf("PWM 0 to 0\n");
    pwm_set(PWM_DEV(0), 0, 0);
    xtimer_sleep(2);
    printf("PWM 0 to 255\n");
    pwm_set(PWM_DEV(0), 0, 255);
  }
  return 0;
}
```
And add in periph_conf.h:
```
/**
 * @name PWM configuration
 * @{
 */
#define PWM_0_EN 1
 
#if PWM_0_EN
/* PWM0 channels */
static const pwm_conf_chan_t pwm_chan0_config[] = {
    /* GPIO pin, MUX value, TCC channel */
    {GPIO_PIN(PA, 9), GPIO_MUX_F, 3},
};
#endif
 
/* PWM device configuration */
static const pwm_conf_t pwm_config[] = {
#if PWM_0_EN
    {
        .tim = TCC_CONFIG(TCC1),
        .chan = pwm_chan0_config,
        .chan_numof = ARRAY_SIZE(pwm_chan0_config),
        .gclk_src = SAM0_GCLK_TIMER,
    },
#endif
};
 
/* number of devices that are actually defined */
#define PWM_NUMOF ARRAY_SIZE(pwm_config)
/** @} */
 ```
 
 I try to play with Waveform Extension, OTMX and WEXCTRL without success

maribu, 2023-01-04:
This seems to be an issue with the application rather than RIOT, as it does work fine for @dylad. And without further input from the reporter, it is difficult to aid debugging. Hence, I close this for now.

Please re-open if further help is needed :)

closed 2023-01-04 by maribu

biboc, 2023-01-23:
I've finally got an answer from Microchip/Atmel
It is a problem of the board and the datasheet

> 
> The old SAMR30 datasheet (http://ww1.microchip.com/downloads/en/DeviceDoc/70005303B.pdf) mentions TCC1 wo3 channel for pin PA09
> 
> whereas in the revised latest datasheet it shows that number of TCC in SAMR30 is reduced and also FECTRL functionality is assigned in mux F for PA09 instead of TCC1. This Is the reason we don't see waveform output for TCC1 in PA09 in SAMR30 MCU.
> 
>  https://ww1.microchip.com/downloads/aemDocuments/documents/OTH/ProductDocuments/DataSheets/IEEE-802.15.4-Sub-GHz-System-in-Package-Datasheet-DS70005303C.pdf#page=1154
> 
>  For using TCC1 , kindly select other configurable pins from latest datasheet . Thanks for bringing this to our attention

So simply, we can't use PWM PA09 with TCC1