RIOT-OS/RIOT #18987: CI: can_fast_ci_run.py misclassifying sub unittests

kaspar030, 2022-11-28:
#### Description

From #18950 

"can_fast_ci_build.py correctly identified just tests/unittests/tests-core as changed, but the build system doesn't understand APPS=tests/unittests/tests-core."

#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

kaspar030, 2022-11-28:
maybe hard-replacing "tests/unittests/*" with "tests/unittests" is a sensible work-around?

MrKevinWeiss, 2022-11-30:
Also occurred on #18734

MrKevinWeiss, 2022-11-30:
I suppose another workaround would be unittest changes should have the full build label?

kaspar030, 2022-11-30:
> I suppose another workaround would be unittest changes should have the full build label?

yup, but is there a benefit? why not just build all unittests?

MrKevinWeiss, 2022-11-30:
> yup, but is there a benefit? why not just build all unittests?

If there is no fix already in, but you promptly supplied one 👍

PR #18996 (2022-11-30): CI: can_fast_ci_build.py unittests fix

### Contribution description

This PR should fix #18987.

I first started with some sophisticated regex based generic replacement / mapping functionality, then decided to go simple...

The issue is that can_fast_ci_run.py *correctly* classifies any change to `tests/unittests/tests-$foo` to be a change to that module, but unittests are special in our build system - they can be built as one huge binary (`tests/unittests`), or individually (`tests/unittests/tests-*`). The latter are not built by CI and are not listed when doing `make info-applications`.

## [... cut]
closed 2022-11-30 by kaspar030