RIOT-OS/RIOT #19000: SoftwareSerial usage in Arduino UNO

mugekuskon, 2022-12-01:
Hello,

I've implemented a code using some libraries such as SoftwareSerial and NewPing in my arduino code. I have named my .ino file to .sketch and run the commands below:

`make BOARD=arduino-uno`
`make BOARD=arduino-uno flash`

No errors have occured except some warnings but RIOT is flashed in my board successfully. Unfortunately, when I have ran my code in Arduino IDE, it was working (connected first to wifi, then send some sensor data to the Thingspeak cloud), however after flashing it, no data is going to Thingspeak anymore. I would be glad if you help me. Thank you in advance.

Sincerely,

kaspar030, 2022-12-02:
Happy 19k everyone!!

trail-coffee, 2022-12-03:
@mugekuskon can you post any code (use the attach feature on comment)? do you have any serial information coming to terminal when you do `make BOARD=arduino-uno term`?

I'm not sure if this is hacky, but on my SAMD21 arduino board (so not ATMEGA like yours), it helps if i put a `ztimer_sleep(ZTIMER_MSEC, 5000)` before anything is printed to serial and received by py-term.

mugekuskon, 2022-12-05:
Of course, thank you. I have attached it.  No information is coming to the serial terminal unfortunately. I am not sure if RIOT has wifi support? Do you have any idea about it? Yes, this is only arduino-uno no wifi is present, I have connected it to esp8266-01 wifi module. Thanks a lot in advance.

[arduino-uno-esp8266.ino](https://github.com/RIOT-OS/RIOT/files/10153920/arduino-uno-esp8266.txt)

trail-coffee, 2022-12-05:
Can you attach what you see in terminal when building? Should basically be a list of all the modules compiled from RIOT.

You might want to delete the api_key in that attachment as well (not sure if bots are scraping GitHub for people's info).

maribu, 2023-05-17:
Since the original reporter is unresponsive, I'm closing this issue. Please feel free to reopen.

If anyone stumbles upon this, I have the following advice:

1. Networking with the Arduino UNO is not possible with RIOT, as RIOT implements the network stack on the target device. Arduino is able to provide network connectivity on the Arduino UNO by moving all the heavy lifting to the ESP.
    - It would be possible to implement some high level SOCK API based network integration that may work with the Arduino UNO
    - However, it makes technically little sense to add an ATmega328P as "main controller" to an ESP8266 / ESP32 as "network controller". The ESP8266 / ESP32 has more spare resources not needed for network than any ATmega MCU can ever dream of. Adding an ATmega328P to a setup that already has an ESP8266 / ESP32 limits the application, increases the hardware cost, adds another item to power, etc.
    - Instead, on should just flash RIOT on the ESP directly and use that for both network connectivity and application logic
2. It is likely that the SoftwareSerial didn't work correctly
    - RIOT assumes that all hardware access goes through RIOT's API, and the application doesn't directly interact with peripheral registers
    - Some Arduino libs directly interact with peripheral registers
    - RIOT provides in the header `serialport.hpp` a compatibility wrapper for Arduino's SerialPort API that should just work. Note that when this is multiplexed with stdio, the symbol rate (often misnamed baudrate) settings are not applied and the stdio backend. In fact, this would allow using SerialPort with stdio backends such as `stdio_rtt` or `stdio_cdc_acm` where a symbol rate makes no sense anyway.

closed 2023-05-17 by maribu