RIOT-OS/RIOT #19025: SPI access in LoRa-e5 and im880b?

jmfriedt, 2022-12-08:
When compiling the ``tests/periph_spi`` example for LoRa-e5 board or im880b, the
execution of
```
> init 1 0 0
Trying to initialize SPI_DEV(1): mode: 0, clk: 0, cs_port: 0, cs_pin: 0
Note: Failed assertion (crash) means configuration not supported
```
in the provided shell fails, preventing sending messages on the MOSI and CLK pins. This
used to work with tag 6bf6b6be6c4723b49f62550a35111e57b7426aa4 where the expected result is
```
> init 1 0 0
SPI_DEV(1) initialized: mode: 0, clk: 0, cs_port: 0, cs_pin: 0
> 
```
and expected signals are observed on the oscilloscope on MOSI and CLK pins. Has the API changed to initialize the SPI bus?
Thanks

jdavid, 2022-12-26:
I can reproduce the issue with the LoRa-E5 board. But I could not test commit 6bf6b6b since at that time `boards/lora-e5-dev` did not exist yet.

@jmfriedt have you done a git bisect to identify with which commit it stopped working?
If you could do that I will try to do a pull request.

benpicco, 2023-01-02:
> This used to work with tag https://github.com/RIOT-OS/RIOT/commit/6bf6b6be6c4723b49f62550a35111e57b7426aa4 

That's a good starting point!
Can you do a 

    git bisect start
    git bisect bad master
    git bisect good 6bf6b6be6c4723b49f62550a35111e57b7426aa4 

to find the offending commit?

benpicco, 2023-01-03:
I hope there are not too much of those - you can use `git bisect skip` to try the next commit.

jmfriedt, 2023-01-03:
A lot of different issues seem to overlap. So far I have on the im880b
```
7ddfb3de4222fe0f641cf36f29339632e9f6bcbf fail       Sat Jun 4 11:55:44 2022
48fc63e0ae60aec05c16cf641b486b1affb58114 no display
bd37457e14fa424a85d2a9f0b040928db6c47001 no display
3a8f74da0cb6ca20266fd2a801c7063e2417fbdc no display Wed Nov 4 23:00:51 2020
4034f9616983d8c653ded05476e93ee5525c6320 OK         Wed May 6  23:46:43 2020
3e8d6388f8706ce15fc8d81ea440fb1555da96b6 OK         Fri Aug 21 22:20:42 2020
daecd9c607d766a3f269d0bde389fbb19e250b85 OK         Tue Aug 25 16:01:25 2020
```
with ``fail`` meaning the SPI port will not initialize, ``no display`` meaning no output on the RS232 port (either port initialization failure or clock initialization failure) and OK meaning SPI port initialization functional. I am not sure how to separate the lack of display from the non-functional SPI !

kaspar030, 2023-01-09:
> Part of the issue is that RIOT developers are providing patches that break the whole system until the last update is provided, so the first step of bisect already fails with

Do you mean that in a PR, the individual commits are not necessarily buildable?
That's a trade-off we chose to make.
You can bisect at merge commit granularity using `git bisect --first-parent ...`

jdavid, 2023-01-13:
Yes it looks like a new warning in GCC 12.

I think this should work, to ignore _address_ warnings:

    CFLAGS="-Wno-error=address" BOARD=im880b  make -C tests/periph_spi

jmfriedt, 2023-01-15:
Thank you all for your help. I reached a functional version with the following bisect sequence
```
174ee0175fc6cff42de72bd82a5f864dbb0f636c no display Thu Sep 9 13:01:49 2021
582f8d84303054e0ae8228b12199d05a02607581 no display Wed Jan 13 13:28:06 2021
f46572a3085090173a00ccd7185f3df0940dda39 no display Tue Oct 27 10:32:18 2020
f862166cbbb7e7f7b2c70971b168036d7badecb7 no display Thu Sep 24 14:20:26 2020
ba67f6aa274a7ecad8c56d543e089f6d21efd5d4 good       Thu Sep 3 22:09:27 2020
```
which I am afraid is conflicting with https://github.com/RIOT-OS/RIOT/pull/18295 when multiple im880b boards are just not communicating over RS232 after reset (I was given a dozen im880b boards with half working through RS232/stm32flash programming and the other half working when programmed by ST-Link/openocd ... we suspect clock initialization issues but have not solved this problem either, preventing the identification of the faulty SPI CS# definition on this board I am afraid).

Thanks.

jdavid, 2023-01-16:
If I follow dc5631e8 was the last good commit, then 195b7e6 (PR #14945) broke RS232. Later this was fixed, but SPI didn't work. So we cannot identify which merge broke SPI. Maybe can we narrow it? Which was the first merge where RS232 started working again?

jmfriedt, 2023-01-16:
OK I give up on the im880b, too many issues at the same time. I just stumbled upon a ST B-L072Z-LRWAN1 LoRa discovery whose communication remains functional to this day. By compiling with
```
CFLAGS="-Wno-error=address" BOARD=b-l072z-lrwan1  make -C tests/periph_spi
```
and transfering the binary file through the mass storage interface, the following bisect sequence
```
master                                         bad
f7bed004ff5529a84fdc63e11ebc5b91e09a4327 bad  Sun Nov 7 07:52:16 2021
fbe6748c57a72c04c081da0d834bd44a0e1eeafb good Thu Feb 18 09:24:33 2021
08671255bea508fb6de7a752c22509cb1eb048ce good Wed Jun 23 16:30:17 2021
174ee0175fc6cff42de72bd82a5f864dbb0f636c bad  Thu Sep 9 13:01:49 2021
6e0c8db99f9065c06d9455282bca9efca4471ca9 good Sun Aug 8 02:28:12 2021
149de73160f68f80257f7db57657e267ad5c98fa good Thu Aug 26 14:37:58 2021
2bf1202fb9812c6bc8a7aa63720b17c828d944ad bad  Thu Sep 2 20:54:01 2021
db16cafe045627e4fab337dbf526ae0d22843455 good Wed Sep 1 11:58:00 2021
65d717f5a082a6fb8a20eb43032b816103e7b3c2 bad  Thu Sep 2 08:54:53 2021 +0200
df205bc4526aa2d1bbf52594ec22d96b96ed4359 good Wed Sep 1 22:11:39 2021 +0300
a1cbcc9edef94c46ccb8a35615049a0e61d80868 bad  Thu Sep 2 08:50:56 2021 +0200
550565f05250876c5c64fda927602a341e283320 good Wed Sep 1 22:28:14 2021 +0200
```
where "good" means the SPI ``init 1 0 0`` is acknowledged and bad means I get
```
> init 1 0 0
Trying to initialize SPI_DEV(1): mode: 0, clk: 0, cs_port: 0, cs_pin: 0
Note: Failed assertion (crash) means configuration not supported
```
the final answer is
```
a1cbcc9edef94c46ccb8a35615049a0e61d80868 is the first bad commit
commit a1cbcc9edef94c46ccb8a35615049a0e61d80868
Merge: 550565f052 2efc50be97
Author: Francisco <femolina@uc.cl>
Date:   Thu Sep 2 08:50:56 2021 +0200

    Merge pull request #15902 from maribu/spi-api-change-1
    
    drivers/periph_spi: let spi_acquire return void
```

jdavid, 2023-01-19:
Reading the source code I realise that the code actually works, the following message is merely informative:

    Note: Failed assertion (crash) means configuration not supported

This is the source code:

```C
        printf("Trying to initialize SPI_DEV(%i): mode: %i, clk: %i, cs_port: %i, cs_pin: %i\n",
               dev, mode, clk, port, pin);
        puts("Note: Failed assertion (crash) means configuration not supported");
        spi_acquire(spiconf.dev, spiconf.cs, spiconf.mode, spiconf.clk);
        spi_release(spiconf.dev);
```

I think there should be another `puts(..)` at the end saying the operation was successful, and the note should rephrased, something like:

    (if below the program crashes with a failed assertion, then it means the configuration is not supported)

I'll do a pull request.

However, the fact that the test pass doesn't mean that SPI works (right?). In the original description @jmfriedt says:

> and expected signals are observed on the oscilloscope on MOSI and CLK pins.

I don't have an oscilloscope to test. Then, how could I verify whether SPI really works or not?

(For context what I want is to fix issue #18975 , thought that may be an SPI problem or something else).

benpicco, 2023-01-19:
> However, the fact that the test pass doesn't mean that SPI works (right?)

The `bench` command only checks how long individual SPI operations take.
If you want to test send and receive, connect MISO and MOSI with a Jumper/Wire and use the `send` command. You should receive the same bytes that you are sending.

This does of course not check if the clock signal has the correct frequency.

jdavid, 2023-01-19:
Well, I can confirm SPI works in the lora-e5 board, I got the sdcard working, see comment https://github.com/RIOT-OS/RIOT/issues/18975#issuecomment-1396913765

jmfriedt, 2023-01-19:
Connecting as loopback MOSI to MISO does not return the transmitted byte using ``send`` after either ``init 1 0 0`` or ``init 0 0 0`` (nor do signals appear on the scope) i.e either using SPI0 or SPI1.

maribu, 2023-05-17:
Since @jdavid got SPI according to https://github.com/RIOT-OS/RIOT/issues/18975 running on the lore-e5 board and I got it working on the im880b board, I'd say this can be closed.

Please feel free to re-open if you disagree. Or to post questions here if you need assistance.

closed 2023-05-17 by maribu