RIOT-OS/RIOT #19070: NimBLE over HCI UART

trail-coffee, 2022-12-21:
#### Description
I would like to be able to use the NimBLE host only stack over HCI UART H4 to communicate to an ESP32 Bluetooth and Wi-Fi chip.

### Useful links
Basing my assumption that this doesn't exist on NimBLE maintainer at RIOT convention saying something like "future: more host support" and me not being able to figure out how to turn on nimble host and set the uart.

https://youtu.be/gK_OPeBs6dQ

Citrullin, 2022-12-22:
How about the other way around? Upgrading the ESP32 Nimble stack to support IPv6 over BLE? 
So, you just use UDP/TCP and even have multi-hop. It's just more of a future proof approach than using HCI UART.

trail-coffee, 2022-12-22:
> How about the other way around? Upgrading the ESP32 Nimble stack to support IPv6 over BLE? 
> 
Sounds good to me, I just want an easy way to hook up to an app and an Xbox controller.

Wasn't everything supposed to be IPv6 ~10 years ago? 😆 I still don't get an IPv6 from my ISP...

trail-coffee, 2023-01-09:
Didn't realize it wasn't implemented in nimble, there's a pending [PR](https://github.com/apache/mynewt-nimble/pull/1399) so it's coming.

Citrullin, 2023-01-17:
> Didn't realize it wasn't implemented in nimble, there's a pending [PR](https://github.com/apache/mynewt-nimble/pull/1399) so it's coming.

Bump the version  in RIOT once it is there, make a PR, link it to this issue, get it merged and this is done :)

Citrullin, 2023-01-17:
> Didn't realize it wasn't implemented in nimble, there's a pending [PR](https://github.com/apache/mynewt-nimble/pull/1399) so it's coming.

That's the wrong NimBLE though. The [ESP32 has a fork of NimBLE](https://github.com/espressif/esp-nimble). So, you have to get them to merge this into their fork as well. Open source is hard, I know.

trail-coffee, 2023-01-17:
> That's the wrong NimBLE though. The [ESP32 has a fork of NimBLE](https://github.com/espressif/esp-nimble). So, you have to get them to merge this into their fork as well. Open source is hard, I know.

NimBLE on ESP32 already has HCI over uart controller, I just need the host stack on my SAMD21 Arduino (host) to talk to the ESP32 (controller). Not sure how hard dealing with espressif is, they do like making huge changes in their idf that ruin all your existing code, so maybe small changes would be a small ask.