RIOT-OS/RIOT #19117: gnrc_sixlowpan_frag: broken if destination is multicast (only on native?)

benpicco, 2023-01-10:
#### Description

When sending datagrams larger than the L2 MTU, 6LoWPAN fragmentation should transparently take care of splitting the datagram into multiple frames and re-assembling them.

While this works well for unicast, multicast datagrams are not correctly re-assembled.

#### Steps to reproduce the issue

Start two 6lo nodes.
On `native` this can be done by starting `examples/gnrc_networking` and `examples/gnrc_border_router`. While no border router is needed for this, the build system will take care of connecting everything if the border router example is used.

    make -C examples/gnrc_border_router all term
    make -C examples/gnrc_networking USE_ZEP=1 all term

#### Expected results

 Fragmentation works with both unicast and multicast.

#### Actual results

##### Unicast fragmentation works

```
> ping -s 128 fe80::ac46:21cb:1370:fa9a%7
136 bytes from fe80::ac46:21cb:1370:fa9a%7: icmp_seq=0 ttl=64 rssi=0 dBm time=0.561 ms
136 bytes from fe80::ac46:21cb:1370:fa9a%7: icmp_seq=1 ttl=64 rssi=0 dBm time=1.538 ms
136 bytes from fe80::ac46:21cb:1370:fa9a%7: icmp_seq=2 ttl=64 rssi=0 dBm time=1.487 ms

--- fe80::ac46:21cb:1370:fa9a%7 PING statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.561/1.195/1.538 ms
```

#### Multicast ping works

```
> ping ff02::1%7
12 bytes from fe80::ac46:21cb:1370:fa9a%7: icmp_seq=0 ttl=64 rssi=0 dBm time=0.730 ms
12 bytes from fe80::ac46:21cb:1370:fa9a%7: icmp_seq=1 ttl=64 rssi=0 dBm time=0.768 ms
12 bytes from fe80::ac46:21cb:1370:fa9a%7: icmp_seq=2 ttl=64 rssi=0 dBm time=0.802 ms

--- ff02::1%7 PING statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 0.730/0.766/0.802 ms
```

#### Multicast fragmentation does not work

```
> ping -s 128 ff02::1%7
ping -s 128 ff02::1%7

--- ff02::1%7 PING statistics ---
3 packets transmitted, 0 packets received, 100% packet loss
```

#### Versions
RIOT master

fabian18, 2023-01-10:
> (only on native?)

For two boards, it works.
```
2023-01-10 11:19:14,441 # ping -s 128 ff02::1%7
2023-01-10 11:19:14,471 # 136 bytes from fe80::4c49:6f54:4f76:101%7: icmp_seq=0 ttl=64 rssi=-31 dBm time=22.615 ms
2023-01-10 11:19:15,460 # 136 bytes from fe80::4c49:6f54:4f76:101%7: icmp_seq=1 ttl=64 rssi=-31 dBm time=20.818 ms
2023-01-10 11:19:16,451 # 136 bytes from fe80::4c49:6f54:4f76:101%7: icmp_seq=2 ttl=64 rssi=-31 dBm time=20.341 ms
2023-01-10 11:19:16,452 # 
2023-01-10 11:19:16,454 # --- ff02::1%7 PING statistics ---
2023-01-10 11:19:16,459 # 3 packets transmitted, 3 packets received, 0% packet loss
2023-01-10 11:19:16,464 # round-trip min/avg/max = 20.341/21.258/22.615 ms
```
zipped pcap: [6lo.zip](https://github.com/RIOT-OS/RIOT/files/10381596/6lo.zip)
Except that Wireshark shows bad FCS for me but that's nothing new.

miri64, 2023-01-16:
Since you are apparently using ZEP quite successfully for testing, does it make sense to define some release specs for it so that these kinds of errors are caught early(er)?

benpicco, 2023-01-16:
Sure! Since ZEP/native is cheap to run, it can just be a normal test that is run as part of CI as it is done with e.g `tests/gnrc_rpl` and `tests/gcoap_fileserver`.

miri64, 2023-01-18:
> Sure! Since ZEP/native is cheap to run, it can just be a normal test that is run as part of CI as it is done with e.g `tests/gnrc_rpl` and `tests/gcoap_fileserver`.

Looking forward to your PR against the Release-Spec repo :-).

closed 2023-01-19 by bors[bot]