RIOT-OS/RIOT #19139: tests/congure_*: tests broken

kaspar030, 2023-01-13:
#### Description

... at least on native and nrf52840dk, found while release testing.

#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

miri64, 2023-01-13:
Huh? Can you give more details? At least the nrf52840dk issue should have been fixed with https://github.com/RIOT-OS/RIOT/pull/19128

kaspar030, 2023-01-13:
> Huh? Can you give more details? At least the nrf52840dk issue should have been fixed with #19128

Sorry, I missed that PR.

kaspar030, 2023-01-13:
here's the backport: https://github.com/RIOT-OS/RIOT/pull/19143

miri64, 2023-01-13:
Do you have a link where those test failed for `native`?

miri64, 2023-01-13:
Ran the tests on my machine: no issues found (except for some deprecation warnings that need fixing)

```
$ for test in tests/congure_*/; do QUIETER=1 make --no-print-directory -C ${test} -j flash test; done
Building application "tests_congure_abe" for "native" with MCU "native".

   text	   data	    bss	    dec	    hex	filename
  42008	    404	  47992	  90404	  16124	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/congure_abe/bin/native/tests_congure_abe.elf
.......
----------------------------------------------------------------------
Ran 7 tests in 13.819s

OK
Building application "tests_congure_quic" for "native" with MCU "native".

   text	   data	    bss	    dec	    hex	filename
  49402	    444	  48136	  97982	  17ebe	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/congure_quic/bin/native/tests_congure_quic.elf
..../usr/lib64/python3.11/unittest/case.py:678: DeprecationWarning: It is deprecated to return a value that is not None from a test case (<bound method TestCongUREQUICDefault.test_on_packet_ack_no_rtt_sample of <__main__.TestCongUREQUICDefault testMethod=test_on_packet_ack_no_rtt_sample>>)
  return self.run(*args, **kwds)
...................../usr/lib64/python3.11/unittest/case.py:678: DeprecationWarning: It is deprecated to return a value that is not None from a test case (<bound method TestCongUREQUICDefault.test_on_packet_ack_no_rtt_sample of <__main__.TestCongUREQUICEventCb testMethod=test_on_packet_ack_no_rtt_sample>>)
  return self.run(*args, **kwds)
.................
----------------------------------------------------------------------
Ran 42 tests in 48.100s

OK
Building application "tests_congure_reno" for "native" with MCU "native".

   text	   data	    bss	    dec	    hex	filename
  43712	    416	  48536	  92664	  169f8	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/congure_reno/bin/native/tests_congure_reno.elf
........................
----------------------------------------------------------------------
Ran 24 tests in 30.001s

OK
/home/mlenders/Repositories/RIOT-OS/RIOT/makefiles/kconfig.mk:118: Warning! SHOULD_RUN_KCONFIG is not set but a previous configuration file was detected (did you run `make menuconfig`?). Kconfig will run regardless.
/home/mlenders/Repositories/RIOT-OS/RIOT/makefiles/kconfig.mk:118: Warning! SHOULD_RUN_KCONFIG is not set but a previous configuration file was detected (did you run `make menuconfig`?). Kconfig will run regardless.
Building application "tests_congure_test" for "native" with MCU "native".

   text	   data	    bss	    dec	    hex	filename
  42946	    420	  48056	  91422	  1651e	/home/mlenders/Repositories/RIOT-OS/RIOT/tests/congure_test/bin/native/tests_congure_test.elf
.................................................
----------------------------------------------------------------------
Ran 49 tests in 23.716s

OK
$ git show --oneline
88bdbcc1ba (HEAD -> 2023.01-branch, upstream/2023.01-branch) Merge #19135
```

miri64, 2023-01-16:
Ping @kaspar030, can this be closed?

kaspar030, 2023-01-16:
_probably_, I'd like to carve the RC2 and see if the manual compile_and_test_for_board runs succeed.

kaspar030, 2023-01-16:
I still have test failures for `tests/congure_abe` and `tests/congure_test`, compiled with compile_and_test_for_board.py and `BUILD_IN_DOCKER=1`, on nrf52840dk.

https://gist.github.com/kaspar030/0df5810bf0574c32160a4175b91f13ba

miri64, 2023-01-17:
Still unable to reproduce on `2023.01-branch` (I did some more test runs without `pytest` and without either docker and the script with the same results)...

```console
$ for i in $(seq 10); do echo "Run $(( i ))"; TEST_EXECUTOR=pytest TEST_EXECUTOR_FLAGS="-vv -s" BUILD_IN_DOCKER=1 ./dist/tools/compile_and_test_for_board/compile_and_test_for_board.py --applications "tests/congure_abe tests/congure_test" . nrf52840dk || break; done
Run 1
INFO:nrf52840dk:Saving toolchain
INFO:nrf52840dk.tests/congure_abe:Board supported: True
INFO:nrf52840dk.tests/congure_abe:Board has enough memory: True
INFO:nrf52840dk.tests/congure_abe:Application has test: True
INFO:nrf52840dk.tests/congure_abe:Run compilation
INFO:nrf52840dk.tests/congure_abe:Run test
INFO:nrf52840dk.tests/congure_abe:Run test.flash
INFO:nrf52840dk.tests/congure_abe:Success
INFO:nrf52840dk.tests/congure_test:Board supported: True
INFO:nrf52840dk.tests/congure_test:Board has enough memory: True
INFO:nrf52840dk.tests/congure_test:Application has test: True
INFO:nrf52840dk.tests/congure_test:Run compilation
INFO:nrf52840dk.tests/congure_test:Run test
INFO:nrf52840dk.tests/congure_test:Run test.flash
INFO:nrf52840dk.tests/congure_test:Success
INFO:nrf52840dk:Tests successful
Run 2
INFO:nrf52840dk:Saving toolchain
INFO:nrf52840dk.tests/congure_abe:Board supported: True
INFO:nrf52840dk.tests/congure_abe:Board has enough memory: True
INFO:nrf52840dk.tests/congure_abe:Application has test: True
INFO:nrf52840dk.tests/congure_abe:Run compilation
INFO:nrf52840dk.tests/congure_abe:Run test
INFO:nrf52840dk.tests/congure_abe:Run test.flash
INFO:nrf52840dk.tests/congure_abe:Success
INFO:nrf52840dk.tests/congure_test:Board supported: True
INFO:nrf52840dk.tests/congure_test:Board has enough memory: True
INFO:nrf52840dk.tests/congure_test:Application has test: True
INFO:nrf52840dk.tests/congure_test:Run compilation
INFO:nrf52840dk.tests/congure_test:Run test
INFO:nrf52840dk.tests/congure_test:Run test.flash
INFO:nrf52840dk.tests/congure_test:Success
INFO:nrf52840dk:Tests successful
Run 3
INFO:nrf52840dk:Saving toolchain
INFO:nrf52840dk.tests/congure_abe:Board supported: True
INFO:nrf52840dk.tests/congure_abe:Board has enough memory: True
INFO:nrf52840dk.tests/congure_abe:Application has test: True
INFO:nrf52840dk.tests/congure_abe:Run compilation
INFO:nrf52840dk.tests/congure_abe:Run test
INFO:nrf52840dk.tests/congure_abe:Run test.flash
INFO:nrf52840dk.tests/congure_abe:Success
INFO:nrf52840dk.tests/congure_test:Board supported: True
INFO:nrf52840dk.tests/congure_test:Board has enough memory: True
INFO:nrf52840dk.tests/congure_test:Application has test: True
INFO:nrf52840dk.tests/congure_test:Run compilation
INFO:nrf52840dk.tests/congure_test:Run test
INFO:nrf52840dk.tests/congure_test:Run test.flash
INFO:nrf52840dk.tests/congure_test:Success
INFO:nrf52840dk:Tests successful
Run 4
INFO:nrf52840dk:Saving toolchain
INFO:nrf52840dk.tests/congure_abe:Board supported: True
INFO:nrf52840dk.tests/congure_abe:Board has enough memory: True
INFO:nrf52840dk.tests/congure_abe:Application has test: True
INFO:nrf52840dk.tests/congure_abe:Run compilation
INFO:nrf52840dk.tests/congure_abe:Run test
INFO:nrf52840dk.tests/congure_abe:Run test.flash
INFO:nrf52840dk.tests/congure_abe:Success
INFO:nrf52840dk.tests/congure_test:Board supported: True
INFO:nrf52840dk.tests/congure_test:Board has enough memory: True
INFO:nrf52840dk.tests/congure_test:Application has test: True
INFO:nrf52840dk.tests/congure_test:Run compilation
INFO:nrf52840dk.tests/congure_test:Run test
INFO:nrf52840dk.tests/congure_test:Run test.flash
[... 82 lines cut ...]
INFO:nrf52840dk.tests/congure_test:Run compilation
INFO:nrf52840dk.tests/congure_test:Run test
INFO:nrf52840dk.tests/congure_test:Run test.flash
INFO:nrf52840dk.tests/congure_test:Success
INFO:nrf52840dk:Tests successful
Run 10
INFO:nrf52840dk:Saving toolchain
INFO:nrf52840dk.tests/congure_abe:Board supported: True
INFO:nrf52840dk.tests/congure_abe:Board has enough memory: True
INFO:nrf52840dk.tests/congure_abe:Application has test: True
INFO:nrf52840dk.tests/congure_abe:Run compilation
INFO:nrf52840dk.tests/congure_abe:Run test
INFO:nrf52840dk.tests/congure_abe:Run test.flash
INFO:nrf52840dk.tests/congure_abe:Success
INFO:nrf52840dk.tests/congure_test:Board supported: True
INFO:nrf52840dk.tests/congure_test:Board has enough memory: True
INFO:nrf52840dk.tests/congure_test:Application has test: True
INFO:nrf52840dk.tests/congure_test:Run compilation
INFO:nrf52840dk.tests/congure_test:Run test
INFO:nrf52840dk.tests/congure_test:Run test.flash
INFO:nrf52840dk.tests/congure_test:Success
INFO:nrf52840dk:Tests successful
```

Seems to be a local issue.

maribu, 2023-05-16:
Looking only at @kaspar030's logs it looks like the test is sending JSON via stdio for python to collect. Given the flakyness of stdio (some lose data when the ringbuf is not polled and fetched faster than it is written to, others use unreliable transport without error handling such as UART), it is quite likely that this translates into flaky tests.

Since I run all tests every now and then and didn't spot any issues, I'd say we can close this. Please reopen, if you disagree.