RIOT-OS/RIOT #19140: rmutex test KO

PhBressy-Cortus, 2023-01-13:
#### Description
rmutex test application cannot provide the expected tests results.
The declared prios is not consistant with the test results. We should have
static const char prios[THREAD_NUMOF] = {THREAD_PRIORITY_MAIN - 1, 4, 5, 2, 3};
but we have in the code source : 
static const char prios[THREAD_NUMOF] = {THREAD_PRIORITY_MAIN - 1, 4, 5, 2, 4};

#### Steps to reproduce the issue
Run the application.

#### Expected results
As described in README.md.
T3 (prio 6, depth 0): trying to lock rmutex now
T4 (prio 4, depth 0): trying to lock rmutex now
T5 (prio 5, depth 0): trying to lock rmutex now
T6 (prio 2, depth 0): trying to lock rmutex now
T7 (prio 3, depth 0): trying to lock rmutex now

#### Actual results
T3 (prio 6, depth 0): trying to lock rmutex now
T4 (prio 4, depth 0): trying to lock rmutex now
T5 (prio 5, depth 0): trying to lock rmutex now
T6 (prio 2, depth 0): trying to lock rmutex now
T7 (prio 4, depth 0): trying to lock rmutex now
The task T7 is prio 4, which cause the following results to be reordered between T4/T7.
Causing a quick compare to fail the test.

#### Versions
All

maribu, 2023-02-22:
Sorry of the long latency. It is indeed intentional that two threads have the same priority here, as RIOT's scheduling is generally predictable and reproducible, except for when external factors (such as a thread being blocked waiting for an IRQ - so e.g. all networking) come into play. So it does indeed make sense to also test for this.

The wording in the README was a bit confusing:

> If two threads have the same priority the lower thread id should acquire the lock.

This is not generally true. Threads with higher priority (that is, lower numeric priority value) will always run before threads with lower priority. But if two threads with the same priority are ready to run, the scheduler will pick whatever thread is in front of the run queue. This just happens to be the thread with the lower id in this app.

PhBressy-Cortus, 2023-02-22:
Sorry, but the latest reported expected results, are showing exactly the same issue.

During the enumeration of the process, described in Read.me
Latest line before the main ...
...
T7 (prio 3, depth 0): trying to lock rmutex now
main: unlocking recursive mutex
...

- T7 is seen as prio 3, but in the source code, T7 is declared as prio 4 !

I agree on the explanations regarding scheduling behaviour, but there is a mismatch beetween code and test Results.
The registered tests results cannot come from the rmutex source code.

- Because 3 != 4

closed 2023-03-15 by bors[bot]