RIOT-OS/RIOT #19264: STM32 family support

benpicco, 2023-02-10:
This gives an overview of STM32 MCUs for which RIOT has at least basic support.
As ST releases new Chips we are not always up to date with the changes needed to support them, but  they typically share a lot of code.

#### High Performance

 - [x] STM32F4
 - [x] STM32F7
 - [ ] STM32H7

### Mainstream
 - [ ] STM32C0
 - [x] STM32F0
 - [x] STM32F1
 - [x] STM32F3
 - [x] STM32G0
 - [x] STM32G4

### Ultra Low Power
 - [x] STM32L0
 - [x] STM32L1
 - [x] STM32L4
 - [ ] STM32L4+
 - [x] STM32L5
 - [x] STM32U5

### Wireless
 - [x] STM32WB
     - [ ] IEEE 802.15.4 radio
     - [ ] BLE radio
 - [x] STM32WL
     - [x] LoRa radio

### Useful links

https://www.st.com/en/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus.html

aabadie, 2023-02-10:
AFAICT, the STM32L4+ are supported, they are just in the same code as STM32L4. They correspond to STM32L4R5xxx for example.

crasbe, 2025-02-20:
I think it would be worth adding the new STM32WB series as well, the only supported one is the WB55 right now.
This is an overview of their current lineup, but for the initial post it's probably enough to add the Series.

STM32WBx5 Series:
- STM32WB55
- STM32WB35
- STM32WB15

STM32WBx0 Series:
- STM32WB50
- STM32WB30
- STM32WB10

STM32WB0x Series:
- STM32WB09
- STM32WB07
- STM32WB06
- STM32WB05

STM32WBA Series:
- STM32WBA50
- STM32WBA52
- STM32WBA54/55
- STM32WBA62
- STM32WBA63
- STM32WBA64/65