RIOT-OS/RIOT #19305: Error `invalid subcommand "write_image erase ..."` when trying to flash `gnrc_border_router` to board

LukasMino, 2023-02-23:
#### Description
I am trying to flash my nrf52840dk board with `gnrc_border_router`, but it always fails.

#### Steps to reproduce the issue
Ubuntu console in WSL 2.0:
```
~$ git clone https://github.com/RIOT-OS/RIOT.git
~$ cd ./examples/gnrc_border_router
~/RIOT/examples/gnrc_border_router$ sudo make clean all flash BOARD=nrf52840dk
```
Without admin priveledges it would not find my board.

#### Expected results
It should flash the board successfully.

#### Actual results
```
~/RIOT/examples/gnrc_border_router$ sudo make clean all flash BOARD=nrf52840dk
make: command: Command not found
Building application "gnrc_border_router" for "nrf52840dk" with MCU "nrf52".

"make" -C /home/.../RIOT/boards/common/init
...
"make" -C /home/.../RIOT/sys/ztimer
   text    data     bss     dec     hex filename
  85072     180   22032  107284   1a314 /home/.../RIOT/examples/gnrc_border_router/bin/nrf52840dk/gnrc_border_router.elf
/home/.../RIOT/dist/tools/openocd/openocd.sh flash /home/mino...gue/RIOT/examples/gnrc_border_router/bin/nrf52840dk/gnrc_border_router.elf
### Flashing Target ###
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Info : No device selected, using first device.
Info : J-Link OB-SAM3U128-V2-NordicSemi compiled Sep 21 2022 09:57:39
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : clock speed 10000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* nrf52.cpu          cortex_m   little nrf52.cpu          running
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000498 msp: 0x20000200
invalid subcommand "write_image erase /home/.../RIOT/examples/gnrc_border_router/bin/nrf52840dk/gnrc_border_router.elf 0"
in procedure 'flash'
make: *** [/home/.../RIOT/examples/gnrc_border_router/../../Makefile.include:849: flash] Error 1
```

I do not know why it says `make: command: Command not found` at the beginning.

#### Versions
Operating system: Windows 11 using WSL 2.0 and Ubuntu 20.04
Board: Nordic Board nrf52840dk

```
~/RIOT$ make print-versions

Operating System Environment
----------------------------
         Operating System: "Ubuntu" "20.04.5 LTS (Focal Fossa)"
                   Kernel: Linux 5.15.79.1-microsoft-standard-WSL2 x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: missing
           msp430-elf-gcc: missing
       riscv-none-elf-gcc: missing
  riscv64-unknown-elf-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.3.0"
        msp430-elf-newlib: missing
    riscv-none-elf-newlib: missing
riscv64-unknown-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: missing
                 cppcheck: missing
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0
                   python: missing
                  python2: missing
                  python3: Python 3.8.10
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

benpicco, 2023-03-06:
How did you get the USB device into WSL2?
What works is to call the Windows binary from withing WSL2, so if you have `JLinkExe.exe` on Windows create a symlink to that with the name `JLinkExe` in a directory in your `PATH`.
Then the Windows tool should be picked up when you set `PROGRAMMER=jlink`.

LukasMino, 2023-03-28:
I actually got it to work, but flashing it only worked using a VM (in my case Workstation Player). But using it inside WSL afterwards is still possible.

@benpicco 
Thank you for your answer :)
I forwarded the USB device from the Windows Shell ([see here](https://learn.microsoft.com/de-de/windows/wsl/connect-usb)). In the end I should have just installed JLink and nrf-tools inside the WSL enviroment and not over Windows but that might work as well (have not tried it).

maribu, 2023-05-10:
> `Open On-Chip Debugger 0.10.0`

That's the issue. That is a prehistoric version (six years old as of now) without a flash driver for nRF5x MCUs. The current stable release of OpenOCD is 0.12.0 - I strongly recommend to use the latest stable release of OpenOCD. See https://openocd.org/pages/getting-openocd.html on how to install that.

I think OpenOCD 0.11.0 would also work with nRF5x, but I'm not sure.

As @benpicco said: Likely the route of least effort is to go with a Windoofs version of J-Link or OpenOCD and call that from within WSL, as you don't have to jump through the USB over IP hoops.

If you need further assistance, feel free to ask here. I'm closing this issue, though, as it is not due to a bug on the RIOT side.