RIOT-OS/RIOT #19319: examples/gnrc_border_router: esp_wifi_init failed with return value 257 on ESP32-C3 with nimble_rpble

benpicco, 2023-02-26:
#### Description

It is possible to do 6LoWPAN over BLE with nimble (see e.g `tests/nimble_rpble_gnrc`) so it makes sense to also use RIOT as a border router between WiFi and BLE with an ESP32.

While combining WiFi and BLE works fine on a classic ESP32, doing the same on ESP32-C3 fails to initialize the WiFi.

#### Steps to reproduce the issue

Apply this patch
```patch
--- a/examples/gnrc_border_router/Makefile.board.dep
+++ b/examples/gnrc_border_router/Makefile.board.dep
@@ -7,7 +7,9 @@ ifeq (,$(filter native,$(BOARD)))
   else ifeq (wifi,$(UPLINK))
     ifneq (,$(filter esp32 esp8266,$(CPU)))
       USEMODULE += esp_wifi
-      USEMODULE += esp_now
+      USEMODULE += bluetil_addr
+      USEMODULE += nimble_rpble
+      NIMBLE_MAX_CONN = 3
     else
       $(error Only esp32 and esp8266 are currently supported)
     endif

```

and flash the border router application on a ESP32-C3 board:

    make -C examples/gnrc_border_router BOARD=esp32c3-devkit UPLINK=wifi all flash term

#### Expected results

Border router between WiFi and BLE (works on ESP32):

```
2023-02-26 23:08:40,261 # Iface  10  HWaddr: 3C:71:BF:9E:13:FD 
2023-02-26 23:08:40,265 #           L2-PDU:1280  MTU:1280  HL:64  RTR  
2023-02-26 23:08:40,267 #           RTR_ADV  6LO  IPHC  
2023-02-26 23:08:40,270 #           Source address length: 6
2023-02-26 23:08:40,273 #           Link type: wireless
2023-02-26 23:08:40,279 #           inet6 addr: fe80::3c71:bfff:fe9e:13fd  scope: link  VAL
2023-02-26 23:08:40,286 #           inet6 addr: 2001:9e8:1412:1efa:3c71:bfff:fe9e:13fd  scope: global  VAL
2023-02-26 23:08:40,288 #           inet6 group: ff02::2
2023-02-26 23:08:40,291 #           inet6 group: ff02::1
2023-02-26 23:08:40,295 #           inet6 group: ff02::1:ff9e:13fd
2023-02-26 23:08:40,297 #           inet6 group: ff02::1a
2023-02-26 23:08:40,298 #           
2023-02-26 23:08:40,304 # Iface  12  HWaddr: 3C:71:BF:9E:13:FC  Channel: 6  Link: up 
2023-02-26 23:08:40,308 #           L2-PDU:1500  MTU:1492  HL:255  RTR  
2023-02-26 23:08:40,311 #           Source address length: 6
2023-02-26 23:08:40,313 #           Link type: wireless
2023-02-26 23:08:40,319 #           inet6 addr: fe80::3e71:bfff:fe9e:13fc  scope: link  VAL
2023-02-26 23:08:40,326 #           inet6 addr: 2001:9e8:1412:1e00:3e71:bfff:fe9e:13fc  scope: global  VAL
2023-02-26 23:08:40,329 #           inet6 group: ff02::2
2023-02-26 23:08:40,331 #           inet6 group: ff02::1
2023-02-26 23:08:40,335 #           inet6 group: ff02::1:ff9e:13fc
2023-02-26 23:08:40,336 #           

2023-02-26 23:08:42,220 # 	pid | name                 | state    Q | pri | stack  ( used) ( free) | base addr  | current     
2023-02-26 23:08:42,228 # 	  1 | sys_evt              | bl mutex _ |   4 |   2616 ( 1184) ( 1432) | 0x3ffb74e8 | 0x3ffb7cc0 
2023-02-26 23:08:42,237 # 	  2 | esp_timer            | sleeping _ |   2 |   3640 (  468) ( 3172) | 0x3ffba6ac | 0x3ffbb310 
2023-02-26 23:08:42,246 # 	  3 | idle                 | pending  Q |  31 |   2048 (  448) ( 1600) | 0x3ffc02d0 | 0x3ffc0910 
2023-02-26 23:08:42,254 # 	  4 | main                 | running  Q |  15 |   3584 ( 1632) ( 1952) | 0x3ffc0ad0 | 0x3ffc13b0 
2023-02-26 23:08:42,263 # 	  5 | 6lo                  | bl rx    _ |  11 |   2048 (  760) ( 1288) | 0x3ffc8618 | 0x3ffc8be0 
2023-02-26 23:08:42,271 # 	  6 | ipv6                 | bl rx    _ |  12 |   2048 (  780) ( 1268) | 0x3ffc4e4c | 0x3ffc5400 
2023-02-26 23:08:42,280 # 	  7 | udp                  | bl rx    _ |  13 |   1024 (  668) (  356) | 0x3ffc8e1c | 0x3ffc8fe0 
2023-02-26 23:08:42,289 # 	  8 | btController         | bl mutex _ |   1 |   3640 ( 1724) ( 1916) | 0x3ffd3b34 | 0x3ffd4730 
2023-02-26 23:08:42,297 # 	  9 | nimble_host          | bl anyfl _ |   1 |   2048 ( 1596) (  452) | 0x3ffcd23c | 0x3ffcd810 
2023-02-26 23:08:42,306 # 	 10 | nimble_netif         | bl anyfl _ |  10 |   2048 (  684) ( 1364) | 0x3ffca0fc | 0x3ffca710 
2023-02-26 23:08:42,314 # 	 11 | wifi                 | bl mutex _ |   1 |   6200 ( 1808) ( 4392) | 0x3ffd5188 | 0x3ffd6780 
2023-02-26 23:08:42,323 # 	 12 | netif-esp-wifi       | bl anyfl _ |  10 |   2048 (  860) ( 1188) | 0x3ffc2aac | 0x3ffc30c0 
2023-02-26 23:08:42,331 # 	 13 | dhcpv6-client        | bl anyfl _ |  13 |   2048 ( 1736) (  312) | 0x3ffc4648 | 0x3ffc4c30 
2023-02-26 23:08:42,340 # 	 14 | RPL                  | bl rx    _ |  13 |   2048 (  724) ( 1324) | 0x3ffc7a34 | 0x3ffc8040 
2023-02-26 23:08:42,346 # 	    | SUM                  |            |     |  37088 (15072) (22016)
```

#### Actual results
2023-02-26 23:05:12,120 # esp_wifi_init failed with return value 257
2023-02-26 23:05:12,121 # *** RIOT kernel panic:
2023-02-26 23:05:12,121 # Unhandled exception
2023-02-26 23:05:12,121 # 
2023-02-26 23:05:12,122 # 	pid | name                 | state    Q | pri | stack  ( used) ( free) | base addr  | current     
2023-02-26 23:05:12,123 # 	  1 | sys_evt              | bl mutex _ |   4 |   2616 (  408) ( 2208) | 0x3fc8b400 | 0x3fc8bca0 
2023-02-26 23:05:12,124 # 	  2 | esp_timer            | sleeping _ |   2 |   3640 (  296) ( 3344) | 0x3fc8be80 | 0x3fc8cb90 
2023-02-26 23:05:12,125 # 	  3 | idle                 | pending  Q |  31 |   2048 (  232) ( 1816) | 0x3fca2118 | 0x3fca2830 
2023-02-26 23:05:12,126 # 	  4 | main                 | pending  Q |  15 |   3584 ( 1448) ( 2136) | 0x3fca2918 | 0x3fca35d0 
2023-02-26 23:05:12,128 # 	  5 | 6lo                  | bl rx    _ |  11 |   2048 (  360) ( 1688) | 0x3fcaa298 | 0x3fcaa930 
2023-02-26 23:05:12,129 # 	  6 | ipv6                 | bl rx    _ |  12 |   2048 (  448) ( 1600) | 0x3fca6b0c | 0x3fca7190 
2023-02-26 23:05:12,130 # 	  7 | udp                  | bl rx    _ |  13 |   1024 (  360) (  664) | 0x3fcaaa98 | 0x3fcaad30 
2023-02-26 23:05:12,131 # 	  8 | btController         | bl mutex _ |   1 |   3640 (  696) ( 2944) | 0x3fcb748c | 0x3fcb8150 
2023-02-26 23:05:12,132 # 	  9 | nimble_host          | bl anyfl _ |   1 |   2048 ( 1048) ( 1000) | 0x3fcaedd4 | 0x3fcaf4b0 
2023-02-26 23:05:12,133 # 	 10 | nimble_netif         | bl anyfl _ |  10 |   2048 (  408) ( 1640) | 0x3fcabca8 | 0x3fcac3a0 
2023-02-26 23:05:12,134 # 	 11 | netif-esp-wifi       | running  Q |  10 |   2048 (  912) ( 1136) | 0x3fca477c | 0x3fca4ec0 
2023-02-26 23:05:12,135 # 	    | SUM                  |            |     |  26792 ( 6616) (20176)
2023-02-26 23:05:12,135 # 
2023-02-26 23:05:12,135 # *** halted.
2023-02-26 23:05:12,135 # 
2023-02-26 23:05:12,136 # Exception @0x4201cce2, cause 0x5: PMP Load access fault
2023-02-26 23:05:12,136 # heap: 67712 (used 54216, free 13496) [bytes]

#### Versions
RIOT master

gschorcht, 2023-02-27:
> 2023-02-26 23:05:12,120 # esp_wifi_init failed with return value 257

Hm, not sure but it could be that WiFi PHY and BLE PHY can't be used at the same time on ESP32-C3. I have to check.

gschorcht, 2026-01-13:
Found the problem, fixed with #21988.

It's always the same: symbols that must also be accessible when the IROM cache is disabled were placed in IROM instead of IRAM or ROM.