RIOT-OS/RIOT #19359: sys/usbus: usbus_cdc_ecm does not work together with usbus_cdc_acm for STM32

gschorcht, 2023-03-07:
#### Description

The combination of USBUS CDC ECM and CDC ACM interface does not work for different STM32 MCUs. It doesn't work for:
- STM32F429ZI FS interface (CID 1100)
- STM32F439ZI FS interface (CID 1200)
- STM32F723IE HS interface (CID 3100)
- STM32F746NG HS interface (CID 2100)

For these MCUs, the enumeration of the CDC ECM interface fails:
```
[224862.292833] usb 1-2.2: Product: nucleo-f439zi
[224862.292835] usb 1-2.2: Manufacturer: RIOT-os.org
[224862.292836] usb 1-2.2: SerialNumber: DC4DB8B40086FC7A
[224862.305285] cdc_acm 1-2.2:1.0: ttyACM1: USB ACM device
[224867.487946] cdc_ether: probe of 1-2.2:1.2 failed with error -22
...
[225333.844310] usb 1-4.1: Product: stm32f723e-disco
[225333.844311] usb 1-4.1: Manufacturer: RIOT-os.org
[225333.844313] usb 1-4.1: SerialNumber: A6BAC4E1B1E0806B
[225333.847348] cdc_acm 1-4.1:1.0: ttyACM1: USB ACM device
[225339.028774] cdc_ether: probe of 1-4.1:1.2 failed with error -32
```
It works for:
- STM32F723IE FS interface (CID 3000)
- STM32F746NG FS interface (CID 2000)
- STM32F767ZI FS interface (CID 2000)

For the FS interfaces it could be related to the 1.x Core version while the problem for the HS interfaces could be HS interface driver. `tests/usbus_cdc_ecm` works for all MCUs (for the HS interface PR #19358 is required).

The problem was already given before the `cpu/stm32/periph/usbdev_otg.c` was moved as modified version to `drivers/usbdev_synopsys_dwc2` (see branch [2022.07](https://github.com/RIOT-OS/RIOT/tree/2022.07-branch)).

#### Steps to reproduce the issue

1. Checkout PR #19358 
2. Use any of the following commands
   ```
    USEMODULE=stdio_cdc_acm BOARD=stm32f429i-disc1 make -j8 -C tests/usbus_cdc_ecm flash
    USEMODULE=stdio_cdc_acm BOARD=nucleo-f439zi make -j8 -C tests/usbus_cdc_ecm flash
    USEMODULE='periph_usbdev_hs_ulpi stdio_cdc_acm' BOARD=stm32f746g-disco make -j8 -C tests/usbus_cdc_ecm/ flash
    USEMODULE='periph_usbdev_hs_utmi stdio_cdc_acm' BOARD=stm32f723e-disco make -j8 -C tests/usbus_cdc_ecm flash
   ```

#### Expected results

#### Actual results

#### Versions

gschorcht, 2023-03-07:
Hm, `assert` is a bit difficult if `stdio_cdc_acm` is used as STDIO. There should be at least an error message.

bergzand, 2023-03-07:
I agree that this should output something, either on the usbdev or preferably on the usbus side. This is imho a more generic issue as this doesn't only occur with the combination of functions mentioned here, but with any combination exceeding the hardware limit.

gschorcht, 2023-03-07:
I also found the reason for these MCUs:

> * STM32F723IE HS interface (CID 3100)
> * STM32F746NG HS interface (CID 2100)

If the `usb_reset_board` module is used, it works for these MCUs. The problem is that the host sends a `SET LINE CODING` request to the CDC ACM interface, but this is not supported if `usb_reset_board` is not enabled. In this case, the control handler of the CDC ACM interface simply returns -1 [here](https://github.com/RIOT-OS/RIOT/blob/777857ae4c57880d67225b9d57687561e83f50be/sys/usb/usbus/cdc/acm/cdc_acm.c#L267), which should result in a STALL, but the enumeration just stops on the host side. Unfortunately, I don't have a logic analyzer that can sample 480 Mbps USB high-speed signals.

But when I use USB FS interface, for example a `nucleo-f767zi` board, the enumeration for the CDC ECM continues only after a 5 seconds timeout if `usb_board_reset` isn't enabled, but continues immediately within milliseconds if `usb_board_reset` is enabled. On the bus the USB core just NAKs the OUT data transfers continuously for the `SET LINE CODING` request until this 5 seconds timeout.

gschorcht, 2023-03-07:
> On the bus the USB core just NAKs the OUT data transfers continuously for the `SET LINE CODING` request until this 5 seconds timeout.

Hm, `STALL` is generate only for IN data transfers [here](https://github.com/RIOT-OS/RIOT/blob/777857ae4c57880d67225b9d57687561e83f50be/sys/usb/usbus/usbus_control.c#L282) but `SET LINE CODING` causes an OUT data transfer. IMHO, we have to set the STALL for either the IN or the OUT endpoint here dependent on the direction of the SETUP.

gschorcht, 2023-03-11:
> Hm, `STALL` is generate only for IN data transfers [here](https://github.com/RIOT-OS/RIOT/blob/777857ae4c57880d67225b9d57687561e83f50be/sys/usb/usbus/usbus_control.c#L282) but `SET LINE CODING` [...]

This problem has been fixed with PR #17086.

gschorcht, 2023-03-11:
> * STM32F723IE HS interface (CID 3100)
> * STM32F746NG HS interface (CID 2100)

The problem for these boards is more complex, see https://github.com/RIOT-OS/RIOT/pull/17085#issuecomment-1464928744.

gschorcht, 2023-03-11:
The problem is fixed for `stm32f429i-disc*` by PR #19380.

closed 2023-03-24 by bors[bot]