RIOT-OS/RIOT #19393: gcoap: gcoap_req_send and related should return negative for errors

bergzand, 2023-03-14:
#### Description

`gcoap_req_send` currently returns non-zero positive on success and either zero or negative for failed transfers. This is error prone for users, even internally (see #19383).

#### Steps to reproduce the issue

Read the documentation

#### Expected results

Clear values for errors

#### Actual results

Both negative and zero are used to signal errors

chrysn, 2023-03-14:
This also affects gcoap_req_send_tl (which AIU is stuck in an API-compatibility-change deprecation cycle that hasn't been completed); are there any other function with weird zero results?

bergzand, 2023-03-15:
> This also affects gcoap_req_send_tl (which AIU is stuck in an API-compatibility-change deprecation cycle that hasn't been completed); are there any other function with weird zero results?

`gcoap_req_send_tl()` is the cause behind this as `gcoap_req_send()` calls it with a specific socket type.

The other one I've found is not necessarily weird, but `gcoap_obs_send` squashes all negative error codes from the internal `_tl_send()` to zero. That's the only one I've found in the gcoap code.