RIOT-OS/RIOT #19421: esp32c3: ziscr extention missing

Teufelchen1, 2023-03-24:
#### Description
When trying to use my brand new esp32c3, I ran into an issue compiling for it.

```/pkg/esp32_sdk/components/hal/esp32c3/include/hal/interrupt_controller_ll.h:36: Error: unrecognized opcode `csrrc s0,mstatus,8', extension `zicsr' required```
Others as well. All related to missing isa extensions.

#### Steps to reproduce the issue

* use 22.04.1-Ubuntu
* go to examples/hello-world (maybe add the esp-sdk pkg)
* run `BOARD=esp32c3-devkit make flash`
* Observe compilation errors

#### Expected results
Plug & Play, Build & Flash

This might be related to #17951
Given that PR, I replaced all occurrences of `-march=rv32imc` with `-march=rv32imac_zicsr`. That worked, I can compile, flash and see the hello world on my tty.

Additionally, I had to disable certain compiler warnings:
```Makefile
CFLAGS += -Wno-enum-conversion
CFLAGS += -Wno-attributes
CFLAGS += -Wno-maybe-uninitialized
```
I also noticed this little message from the booting device:
```
# Calculated: 363da575129f253d18d4fc05a75bcfffe9a08e52786188f5f5d06a74cfd5bad5
# Expected: cb0f9786add2bad1adce930935dcb7a3ff28232451b0d3cd6255a9101fff41fa
# Attempting to boot anyway...
```
Not sure if relevant, as I'm new to esp32c3.

<details>
  <summary>Click me: all rv32imc -> rv32imac_zicsr replacements</summary>

```
 diff --git a/makefiles/arch/riscv.inc.mk b/makefiles/arch/riscv.inc.mk
index 3544b214b1..ed78a2bc05 100644
--- a/makefiles/arch/riscv.inc.mk
+++ b/makefiles/arch/riscv.inc.mk
@@ -44,12 +44,13 @@ ifeq (,$(TARGET_ARCH))
 endif
 
 ifeq ($(TOOLCHAIN),gnu)
-  GCC_DEFAULTS_TO_NEW_RISCV_ISA ?= $(shell echo "typedef int dont_be_pedantic;" | $(TARGET_ARCH)-gcc -march=rv32imac -mabi=ilp32 -misa-spec=2.2 -E - > /dev/null 2>&1 && echo 1 || echo 0)
+  GCC_DEFAULTS_TO_NEW_RISCV_ISA ?= $(shell echo "typedef int dont_be_pedantic;" | $(TARGET_ARCH)-gcc -march=rv32imac_zicsr -mabi=ilp32 -misa-spec=2.2 -E - > /dev/null 2>&1 && echo 1 || echo 0)
 endif
 
+
 GCC_DEFAULTS_TO_NEW_RISCV_ISA ?= 0
 
-CFLAGS_CPU := -march=rv32imac -mabi=ilp32
+CFLAGS_CPU := -march=rv32imac_zicsr -mabi=ilp32
```

```
diff --git a/cpu/esp32/esp-idf/esp_idf_cflags.mk b/cpu/esp32/esp-idf/esp_idf_cflags.mk
index 3e7ac6b896..aafe2ea2da 100644
--- a/cpu/esp32/esp-idf/esp_idf_cflags.mk
+++ b/cpu/esp32/esp-idf/esp_idf_cflags.mk
@@ -35,7 +35,7 @@ CFLAGS += -Wno-cast-align
 ifneq (,$(filter riscv32%,$(TARGET_ARCH)))
   INCLUDES += -I$(ESP32_SDK_DIR)/components/riscv/include
   CFLAGS += -DCONFIG_IDF_TARGET_ARCH_RISCV
-  CFLAGS += -march=rv32imc
+  CFLAGS += -march=rv32imac_zicsr
   CFLAGS += -Wno-error=format=
   CFLAGS += -nostartfiles
   CFLAGS += -Wno-format
```

```
diff --git a/cpu/esp32/bootloader/Makefile b/cpu/esp32/bootloader/Makefile
index dfb4a5df06..b56655cbfe 100644
--- a/cpu/esp32/bootloader/Makefile
+++ b/cpu/esp32/bootloader/Makefile
@@ -167,7 +167,7 @@ CFLAGS = -include '$(RIOTBUILD_CONFIG_HEADER_C)' \
 ifneq (,$(filter riscv32%,$(TARGET_ARCH)))
   INCLUDES += -I$(ESP32_SDK_DIR)/components/riscv/include
   CFLAGS += -DCONFIG_IDF_TARGET_ARCH_RISCV
-  CFLAGS += -march=rv32imc
+  CFLAGS += -march=rv32imac_zicsr
   CFLAGS += -Wno-error=format=
   CFLAGS += -nostartfiles
   CFLAGS += -Wno-format
```

```
diff --git a/cpu/esp32/Makefile.include b/cpu/esp32/Makefile.include
index 956a132e2f..4167204fb0 100644
--- a/cpu/esp32/Makefile.include
+++ b/cpu/esp32/Makefile.include
@@ -206,7 +206,7 @@ CFLAGS += -D_CONST=const
 
 # TODO no relaxation yet
 ifneq (,$(filter riscv%,$(TARGET_ARCH)))
-  CFLAGS += -mno-relax -march=rv32imc -mabi=ilp32 -DRISCV_NO_RELAX
+  CFLAGS += -mno-relax -march=rv32imac_zicsr -mabi=ilp32 -DRISCV_NO_RELAX
 endif
 
 ifneq (,$(filter xtensa%,$(TARGET_ARCH)))
```
</details>

dylad, 2023-04-07:
I think @gschorcht is currently fixing it with #19450
You could give it a try and tell us if it's fix your issue.

gschorcht, 2023-04-08:
Exactly, PR #19450 provides the changes to be able to compile RIOT for ESP32-C3 with the Espressif GCC v12.2 toolchain.

Espressif provides its own RISC-V toolchain and the current master does not support GCC v12.2 but only GCC v8.4 as documented. Using `${RIOTBASE}/dist/tools/esptools/install.sh esp32c3` installs the toolchain version that is supported and in the `PATH` variable with `. ${RIOTBASE}/dist/tools/esptools/export.sh esp32c3`.

maribu, 2023-05-10:
@Teufelchen1: Can you confirm that https://github.com/RIOT-OS/RIOT/pull/19450 solved the issue?

closed 2023-05-11 by Teufelchen1