RIOT-OS/RIOT #19469: 2023.04 release bug tracking

MrKevinWeiss, 2023-04-14:
#### Description

- `compile_and_test_for_board.py` found the following:
  - `BOARD=all`
    - [`tests/pkg_libschc`](https://github.com/RIOT-OS/RIOT/issues/19445)
  - `BOARD=arduino-mega2650`
    - `tests/congure_quic`
    - `tests/congure_test`
    - `tests/pbkdf2`
    - `tests/phydat_unix`
    - `tests/pkg_littlefs2`
    - `tests/pkg_mbedtls`
    - `tests/sys_sema_inv`
  - `BOARD=b-l475e-iot01a`
    - `tests/ztimer_overhead`
  - `BOARD=nrf52840dk`
    - `tests/periph_timer_periodic`
  - `BOARD=samr34-xpro`
    - `tests/dbgpin`
    - `tests/mtd_raw`
    - `tests/periph_rtt`

#### Steps to reproduce the issue
[actions results](https://github.com/RIOT-OS/RIOT/actions/runs/4601443283)

dylad, 2023-04-14:
For `samr34-xpro`, 

-` tests/dbgpin` hangs because of the test default pin (port 0, pin0) -> PA00. On SAML21-based MCU, this pin is used as the input for 32kHz crystal which is use internally to feed the PLL which also supply the core. So we simply need to change the default pin.
- `tests/mtd_raw` MTD initialization seems to fail with errno -5 (Input/Output). Don't know why yet.
- `tests/periph_rtt` Tried on iotlab, cannot reproduce. (I don't own one unfortunately)

MrKevinWeiss, 2023-04-26:
Local tests of RC3 result in:
# Failure Summary
- `frdm-kw41z`: ❌ FAIL
  - tests/mtd_flashpage
  - tests/periph_rtc
  - tests/periph_timer
- `nucleo-l452re`: ❌ FAIL
  - tests/pkg_libschc
- `esp32-wroom-32`: ✅ PASS
- `saml10-xpro`: ❌ FAIL
  - tests/bench_ztimer
  - tests/congure_quic
  - tests/event_periodic_callback
  - tests/event_wait_timeout_ztimer
  - tests/evtimer_mbox
  - tests/evtimer_msg
  - tests/evtimer_underflow
  - tests/gnrc_ipv6_fwd_w_sub
  - tests/gnrc_ipv6_nib
  - tests/gnrc_ipv6_nib_6ln
  - tests/gnrc_lorawan
  - tests/gnrc_lorawan_11
  - tests/gnrc_ndp
  - tests/gnrc_netif_ipv6_wait_for_global_address
  - tests/gnrc_rpl_p2p
  - tests/gnrc_sixlowpan_frag
  - tests/gnrc_sixlowpan_iphc_w_vrb
  - tests/gnrc_sock_async_event
  - tests/gnrc_sock_ip
  - tests/gnrc_sock_neterr
  - tests/gnrc_sock_udp
  - tests/lwip_sock_ip
  - tests/lwip_sock_udp
  - tests/netdev_test
  - tests/netutils
  - tests/periph_rtc
  - tests/periph_rtt
  - tests/periph_rtt_min
  - tests/pkg_libschc
  - tests/pkg_mbedtls
  - tests/pkg_umorse
  - tests/posix_sleep
  - tests/ps_schedstatistics
  - tests/pthread_cooperation
  - tests/shell_lock
  - tests/sys_sched_round_robin
  - tests/trickle
  - tests/ztimer_periodic
  - tests/ztimer_xsec
- `nrf52840dk`: ✅ PASS
- `remote-revb`: ❌ FAIL
  - examples/micropython
  - tests/congure_test
  - tests/mtd_raw
  - tests/pkg_mbedtls
  - tests/unittests
  - tests/pkg_libfixmath
  - tests/pthread_cooperation
- `arduino-mega2560`: ❌ FAIL
  - tests/congure_quic
  - tests/congure_test
  - tests/pbkdf2
  - tests/phydat_unix
  - tests/pkg_littlefs2
  - tests/pkg_mbedtls
  - tests/sys_sema_inv
  - tests/ztimer_overhead
- `nucleo-f429zi`: ✅ PASS

MrKevinWeiss, 2023-04-26:
I suppose I should [share](https://github.com/MrKevinWeiss/riot-board-test-results/tree/master/2023.04-branch-f46d71e1686d9ffea4d591dc7d28b978c2bfd261) like @maribu 
I will be improving this a bit in the future to make it easier to share/combine/present/repeat runs.

benpicco, 2023-04-26:
[b-l475e-iot01a:tests/cpp_ctors](https://github.com/MrKevinWeiss/riot-board-test-results/blob/master/2023.04-branch-f46d71e1686d9ffea4d591dc7d28b978c2bfd261/b-l475e-iot01a/tests/cpp_ctors/test.failed) looks more like a network error

MrKevinWeiss, 2023-04-27:
The local tests I reran with `--incremental` but the iotlabs ones are just run once, so I would assume the flakey tests would show more.

maribu, 2025-11-17:
Indeed, let's retire this issue. It seems a number of issues have indeed been fixed in the meantime and #21882 is more up to date