RIOT-OS/RIOT #19511: nrfutil uses subcommands incompatible with the latest version

chrysn, 2023-04-25:
Nordic updated their nrfutil:
* The [6.x series](https://github.com/NordicSemiconductor/pc-nrfutil) is deprecated.
* The 6.x series is also barely usable any more: It doesn't work with Python 3.11 (trying there would install a way older Python 2 version). The workaround I'm using while I have a Python 3.10 around is `python3 -m virtualenv --python /usr/bin/python3.10 /tmp/nrfutil` -- unlike the builtin venv, virtualenv can create those even for older Python versions.
* The newer 7.x series is available [from their website](https://www.nordicsemi.com/Products/Development-tools/nrf-util) as a binary only; the 6.x series was not Free Software either, but at least source was available.
* The 7.x series doesn't work with our Makefiles:
```
nrfutil pkg generate --hw-version 52 --sd-req 0x00 --application-version 1 --application .../examples/opendsme/bin/nrf52840dongle/opendsme_example.hex .../examples/opendsme/bin/nrf52840dongle/opendsme_example.hex.zip
Error: nrfutil command `pkg` not found. See `nrfutil list` for full list of installed commands, `nrfutil search` for installable commands, and `nrfutil install` for installation of new commands.
                                                                                                          
Caused by:                                                                                                
    Subcommand nrfutil-pkg not found         
```

AFAICT this only affects the nrf52840dongle as it's the only board that can actually use the (non-adafruit, they forked off some time ago) nrfutil.

I have no clear way forward. Might be "ditch nrfutil and focus on riotboot", but it'd be great to have some way to migrate without using old software and without using an external programmer.

aabadie, 2023-04-26:
Why not check from the RIOT build system which version of nrfutil is installed and adapt the command line accordingly ?

chrysn, 2023-04-26:
Probably, but given it's nonfree software I'll only touch it when absolutely necessary. The fact that it writes to ~/.nrfutil without asking doesn't make me any happier about using it, and the fact that it tries to create a file /tmp/nrfutil on self-upgrade (tough look, I have a dir there, may I introduce you to mktmp...) gives me serious doubt about its quality of implementation to the point where I'm not sure I'd support it in RIOT at all.